Ultimaker/Cura #6388: All of a sudden "Open File(s)" freezes Cura 4.2.1

blibenson, 2019-09-20:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

nallath, 2019-09-23:
So it used to work for 4.2.1, but after you installed Cura 4.3 beta both of them stopped working?

fieldOfView, 2019-09-23:
I think this may be specific to the Downloads folder on Windows. 
See https://www.laptopmag.com/articles/slow-windows-downloads-folder

blibenson, 2019-09-23:
So a couple of things. When I empty the download folder it works fine. I then copied all the .stl files back and it worked as usual. Not sure if a particular file was causing the issue.

blibenson, 2019-09-23:
> So it used to work for 4.2.1, but after you installed Cura 4.3 beta both of them stopped working?

Not exactly. It stopped working under 4.2.1 so I tried 4.3 then tried re-installing 4.2.1. Failed both times until I re-created the download folder.

closed 2019-09-23 by blibenson

jay-packet, 2019-10-14:
I've verified an issue.  If Cura's splash/load screen appears on my main monitor ( I have three monitors active) and Cura's interface loads on a different monitor it will freeze. As soon as I disconnected the other two monitors to force Cura to load the splash screen and the main screen on the same monitor it fixed the issue.  No problems loading and no freezing. I'm using Cura 4.3.0