Ultimaker/Cura #6389: Simultaneous bed and extruder heating in preamble

sailorgreg, 2019-09-20:
**Application version**

4.2.1

**Platform**

Windows

**Printer**

Sainsmart Ender3

**Reproduction steps**

Every print

**Actual results**

================================
;Generated with Cura_SteamEngine 4.2.1
T0
M190 S80
M104 S230
M109 S230
M82 ;absolute extrusion mode
============================
I wait for the bed to heat while the extruder sits at "cold" temp 
then the extruder heats while the bed is at full temp.

**Expected results**

============================
;Generated with Cura_SteamEngine 4.2.1
T0
M140 S80
M104 S230
M109 S230
M190 S80
M82 ;absolute extrusion mode
=============================
By adding:
M140 S80    ; the bed starts heating
M104 S230  ; the extruder starts heating
M109 S230  ; we now wait for the extruder temp while BOTH the bed and the extruder are heating
M190 S80    ; if necessary, we wait for the bed to reach target temp

**Additional information**

I didn't know whether to put this into the bug or feature list but it seems pretty minor and could easily be implemented in a minor upgrade in the near future.

diegopradogesto, 2019-09-24:
Please send us a project file (File > Save...) so that we can analyse the specific case for the specific printer.

sailorgreg, 2019-09-24:
I would have thought this was so obvious that an example file was unnecessary but here you go.

 

This is the Gcode that is generated:
M190 S70

M104 S220

M109 S220

And I hack it every time to be:
M140 S70

M104 S220

M109 S220

M190 S70

 

Super simple and super obvious but tiresome to have to do after every slicing. 
If I were programming this, it would take me probably 30-90 seconds to add this simple change but I don't know how the generic program design is put together so it could be much more difficult.

Greg

fieldOfView, 2019-09-24:
> If I were programming this, it would take me probably 30-90 seconds to add this simple change

And if I were programming this, I would stop right here and now considering your request. There is no need to be condescending to people who want to help you.

You can add your own heatup sequence in your start gcode. 
```
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
M190 S{material_bed_temperature_layer_0}
```
Enter the text like this (including the {} brackets), anywhere you want in your start gcode snippet in the Machine Settings dialog.

sailorgreg, 2019-09-25:
> And if I were programming this, I would stop right here and now considering your request. There is no need to be condescending to people who want to help you.
>
> You can add your own heatup sequence [...]

fieldOfView, Sorry about the attitude but was tired after a long day. Thanks for your help. 

Where do you find this type of customization information in the Cura documentation? I was looking for something like this but couldn't find it. Cura seems quite powerful and versatile and I was surprised that I couldn't find it. Thanks again!

fieldOfView, 2019-09-26:
As an external constributor, I can answer questions here. I have no say over Cura documentation.

Ghostkeeper, 2019-09-26:
Ultimaker's website contains the "mainstream" documentation for Cura but doesn't contain this. Ultimaker doesn't spend any effort in documenting this because it's not something that's needed for their printers.

I've started a documentation project of my own by expanding the scope of the [Settings Guide](https://github.com/Ghostkeeper/SettingsGuide) plug-in. However currently it also doesn't document this yet. There's just so many things to write about.