Ultimaker/Cura #6390: Dual Extrusion menu improvements

Christanoid, 2019-09-21:
**Is your feature request related to a problem? Please describe.**
Switching Materials in a Dual Extrusion menu takes a lot of time.

**Describe the solution you'd like**
I'd suggest two changes to Cura in order to improve the workflow of changing around Materials in Cura.
a) Make it so that the Profile selection doesn't have 1 Profile for both extruders but instead that extruders have their own dedicated profile. This way mixing together new Material combinations doesn't require you to create a new Profile and migrate settings from other profiles to the new "mixture" profile
b) Make it so that you can assign a Material to a Profile, this can also be useful for single extruder switching. As I noticed I always have to switch the Material AND the Profile in order for it to be really switched. Now if I'd be able to assign a Material to a Profile and have the Material switch with the profile one of those steps would fall away.

**Affected users and/or printers**
Affected users would be pretty much everyone, because it's a setting everybody has to use and also every printer would be affected

Ghostkeeper, 2019-09-24:
a) I hope I'm understanding you correctly. You are asking for custom profiles to be set per extruder rather than for the entire printer the same. We made it to be the same profile (outwards looking) for the entire printer because then you can store global settings as well, like layer height. And because it makes the interface simpler.

b) Having the custom profile depend on the material that it was made for is something we can reconsider, but it's a bit difficult because that would prevent you from using your profile with different materials. Like if you switch to Colorfab PLA it'll probably be fine with you changing the infill density as well. It also depends on the implementation for a).

AbeFM, 2019-09-24:
I agree with OP - certainly on point 1.

It would be nice if active extruders was in the profile, then. I have a different profile for using the printer with only one nozzle, but it always loads in the prime tower, whose settings I cannot get to but are invalid, so I have to activate the extruder, a bunch of settings change without warning, then after fixing them, and I set it back, I find there's something I missed.

I don't see why that cannot be part of the setting, by the logic you're responding to OP with. Certainly the long standing error of refusing to slice when the non-enabled prime tower is in a bad spot should be fixed. :-)

Ghostkeeper, 2019-09-25:
> I don't see why that cannot be part of the setting, by the logic you're responding to OP with. Certainly the long standing error of refusing to slice when the non-enabled prime tower is in a bad spot should be fixed.

Indeed settings that are hidden should not be preventing you from slicing. That seems to be a bug.

Christanoid, 2019-09-25:
@Ghostkeeper I can see your arguments against my suggestions, but I'd
a) just let extruder 0 be the "lead extruder" that sets the global settings from his profile for all
b) either assign it the other way around then, so that the profile is assigned to the material and only have the profile autoswitch with the material, not the other way around or 
  open a small dialogue box that asks if you want to switch the correlating profile / material that was assigned right away with it.

AbeFM, 2019-11-05:
I agree that it's a pain to pick a new material, then have to set the temperatures for it, the retractions, basically all the things that are material dependent, like Material Printing Temperature and Nozzle Switch Retraction Distance.

I can easily clog a nozzle if I carefully select "PVA" from materials but I was earlier printing with TPU. It DOES change initial printing temperature, but it's not a complete list.

It'd take some thinking to know what to change but I can say it's not right now. :-)