Ultimaker/Cura #6392: [4.2.1] Printing time skips around

timberwolf90, 2019-09-22:
**Application version**

Cura 4.2.1

**Platform**

DxDiag had no errors
OS: Windows 10 Home 64-bit (10.0, Build 17134) (17134.rs4_release.180410-1804)
Processor: Intel(R) Core(TM) i7-5500U CPU @ 2.40GHz (4 CPUs), ~2.4GHz
Graphic Card name: Intel(R) HD Graphics 5500

**Printer**

Geeetech A10M
renamed file to a .zip
[CFFFP_B3-8.3mf.zip](https://github.com/Ultimaker/Cura/files/3639287/CFFFP_B3-8.3mf.zip)

**Reproduction steps**

I looked at my printing time number to see how long the print job was taking and noticed the seconds were not adding up. they would vary up and down and the hours and time were way off base. The project had been on for 3 hours and 28 minutes (according to Geeetech's LCD panel) but Cura was saying 6 hours,  17 minutes, 13 seconds. the seconds fluctuate up and down (13, 14, 15, 13, 12, 14, 15, 16....etc). In the long run the printing time goes down by 5 or so minutes. Never goes up.
Started today (9/21/19)

**Actual results**

**Expected results**

**Additional information**

[cura.log](https://github.com/Ultimaker/Cura/files/3639288/cura.log)

Ghostkeeper, 2019-09-24:
Cura's "Printing Time" in the USB printing monitor is the expected total time to print. It's the printing time so far, added to the estimated remaining time. So if you have a project that had been going on for 3.5 hours and Printing Time says 6.5 hours, that means that you're just past the 50% mark in your print.

This is also why the printing time fluctuates a bit but stays approximately the same. That means that the time estimates are correct. Towards the end of the print it should fluctuate less because the remaining time estimate is lower.

Ghostkeeper, 2019-09-24:
Closing this because it's expected behaviour. But maybe if you know how we can make it more clear you can still give a suggestion.

closed 2019-09-24 by Ghostkeeper