Ultimaker/Cura #6393: QoL Improvements for Cura 4.3

Keichide, 2019-09-22:
So i'm currently using Cura 4.2.1 but was also annoyed by some of the stuff in the previous Versions. 

What i like to see in Cura 4.3 is a bigger brim for the Tree Supports or at least an Option to make i bigger, because seriously you can't call it a brim the way it is right now, that's just a joke.. 

The next thing would be, to let the User change every setting (except the layer height) through Post Processing. Because right now for example i have to relay on a presclied G-Code from someone else just to test different Retract Settings and still have no proof if the same settings are working afterwards with my Cura Profil. 

Another one would be the experimental Bridge Settings.. Did you guys ever try to print a CaliCat with no infill, because you wanted to try out translucent pla? Cura never noticed that there is no infill under the bottom Layers and always tried to print a Concentric pattern instead of printing a Bridge first, there should really be something like a detection build in that changes the pattern if there is no infill under the top layer and change the pattern to lines.

And the last QoL would be the Ironing movement, why is it doing exactly the same movement as on a surface pattern? That's just dumb and ruins the quality of the top surface. Instead it should just go in a Lines pattern non stop from the top left to the bottom right, and rise the nozzle if there is something in the way. This way it will resolve in a much cleaner looking surface, also it would be nice to have an option to raise the nozzle a little bit for the ironing.

smartavionics, 2019-09-22:
> Another one would be the experimental Bridge Settings.. Did you guys ever try to print a CaliCat with no infill, because you wanted to try out translucent pla? Cura never noticed that there is no [...]

This has been addressed (https://github.com/Ultimaker/Cura/pull/5635) but that PR has not yet been integrated into Cura so it won't be in 4.3. When it does, you can expect to see the first skin layer above sparse infill to be printed using the bridge skin settings.

[image]

smartavionics, 2019-09-22:
BTW, if you want to try that today, install one of my experimental builds which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Keichide, 2019-09-22:
> 
> This has been addressed (#5635) but that PR has not yet been integrated into Cura so it won't be in 4.3. When it does, you can expect to see the first skin layer above sparse infill to be printed using the bridge skin settings.
> 
> [image]

Thanks, but i'm not in a hurry right now to use it, because i already used up all of the samples and did the top layers with the Line Pattern. But like i said.. It would be nice to have it in a stable release.

fieldOfView, 2019-09-22:
For future reference, please open one issue per request and fill in the issue template. Otherwise your suggestions will be harder to keep track of, and the likeliness of them getting implemented will be small.

rburema, 2019-09-24:
- 4.3.0 will be out _very_ shortly (we set our sights on 'today'), so none of this is going to be in there.
- Brim situation:  As a workaround, you can 'Enable Support Brim'. Otherwise the brim (as long as 'Brim Only On Outside' is off) should be there. This could be a bug.
- Concerning presliced gcode: it's only the toolpaths, the model isn't saved. For specific settings there are things we could do (some easier, some harder), but it's roughly 500 settings that'd have to have a special 'post-processing' version/implementation if we do them all (and I don't think 'layer height' is the only one that's going to be if not impossible, very hard).
- Ironing request: Could I reword that as 'Ironing Pattern Selection'?

Ghostkeeper, 2019-09-24:
> What i like to see in Cura 4.3 is a bigger brim for the Tree Supports or at least an Option to make i bigger, because seriously you can't call it a brim the way it is right now, that's just a joke..

We know of a bug left in the brim that it sometimes removes contours on the inside of the print even if Brim Only on Outside is disabled.

> Another one would be the experimental Bridge Settings.. Did you guys ever try to print a CaliCat with no infill, because you wanted to try out translucent pla? Cura never noticed that there is no [...]

See https://github.com/Ultimaker/Cura/issues/5127

Keichide, 2019-09-25:
> * Brim situation:  As a workaround, you can 'Enable Support Brim'. Otherwise the brim (as long as 'Brim Only On Outside' is off) should be there. This could be a bug.

You missunderstood me. The Brim is there, but it's only on the inside of the Tree Support, which i think could also be a Bug, because my understanding of the Support Brim Setting is that it should look like the Brim in the "Build Plate Adhesion" section. So there should also be a bit more of the Brim on the Outside of the tree model and not just on the Inside. 

> * Concerning presliced gcode: it's only the toolpaths, the model isn't saved. For specific settings there are things we could do (some easier, some harder), but it's roughly 500 settings that'd have [...]

You already covered most of the Stuff somehow in Cura, with a bit of a Work Around, like changing the Infill for certain parts of the Model to 100% with the use of the Support Blocker. But like i said, there is an Option missing to chancge the Retract Length and speed, to make a proper Retraction Test Tower in Cura. 

> * Ironing request: Could I reword that as 'Ironing Pattern Selection'?
If it would solve the Issue that the Ironing feature is exactly doing the same thing as the top/bottom layer.. Yes. Like i said, the path for it should be more like an Iron and go straight from one side to the other and not function like it's laying another Top/Bottom Layer.

rburema, 2019-09-26:
> You missunderstood me. 

Yes I realised right after. We've made an internal ticket for this (Devs: CURA-6812)

> But like i said, there is an Option missing to change the Retract Length [for post processing]

We've chosen to defer this.

> If it would solve the Issue that the Ironing feature is exactly doing the same thing as the top/bottom layer..

It does, if you have Ironing on, you can change between lines and concentric for ironing specifically.