Ultimaker/Cura #6394: Print all holes walls per layer

FlailAway, 2019-09-22:
**Application version**

4.3.BETA

**Platform**
Mint 19.2, Cinnamon

**Printer**
Any printer

**Reproduction steps**
Printing anything with holes

**Actual results**
Holes are printed one ring at a time

**Expected results**
Print all rings per layer.

**Additional information**
I have a large rectangle with 4 posts, one in each corner. I am using 6-walls and am watching this thing print one-circle for the post outer-wall, go to the next corner and repeat for all 4. 

Start back at the original corner and print the next inner-wall. It takes 6 full trips around the rectangle for the walls of the 6 x 4 posts on one layer.

Can you please do all 6-circular walls at the one corner-stop? Then go to the next corner and do all circles etc. I have "optimize travel," checked, is that causing this? I am stuck in a 8 hour print and cannot re-slice to see. But it looks like at least 1-hours is being taken up dancing around the perimeter.

Other than that, this Beta is looking good.

smartavionics, 2019-09-22:
Try enabling the optimize wall printing order setting. If that doesn't produce the results you want, please save the project file (File -> Save) and zip or rename the .3mf file and attach to this issue. Thanks.

FlailAway, 2019-09-23:
Ahhhhh, OK, that fixed it, thank you.

rburema, 2019-09-24:
Seems fixed for the user, closing the issue.

closed 2019-09-24 by rburema