Ultimaker/Cura #6395: [4.2.1] final "Extruder End G-code" never executed

golfromeo-fr, 2019-09-22:
**Application version**
4.2.1

**Platform**
Win7 SP1 64 bits

**Printer**
Prusa i3 clone

**Reproduction steps**
1/ configuration for 2 extruders with all the "Extruder Start G-code" and "Extruder End G-code" for both extruder
2/ Add a test cube (stl or amf file) on the plate
3/ select the first extruder for the cube
4/ slice the cube
5/ save the file to watch the gcode produced (1 cube / 1 extruder)
6/ duplicate the cube on the plate
7/ select the second extruder for the second cube
8/ slice the cube
9/  save the file to watch the gcode produced (2 cubes / 2 extruder)

**Actual results**
both gcode files (1 or 2 extruders/tools) do not contain the final "Extruder End G-code" so the filament remains loaded 

of course, all the load/unloading are done properly before the final tool is loaded

**Expected results**
the final "Extruder End G-code" is executed so the filament is properly unloaded

**Additional information**
[cura.zip](https://github.com/Ultimaker/Cura/files/3640244/cura.zip)
gcode for the second file
search "LOAD FILAMENT T" you will find the load of T1 is done but not the unload of T1 at the end
[CFFFP_xyzCalibration_cube (2).zip](https://github.com/Ultimaker/Cura/files/3640246/CFFFP_xyzCalibration_cube.2.zip)

Ghostkeeper, 2019-09-24:
Same as https://github.com/Ultimaker/Cura/issues/5629. This is intentional.

Some printers need to do stuff at the extruder switch but not at the end of the print. This is fairly common for wiping the nozzle and priming/unpriming, because these things can be done more thoroughly or not at all at the end of the print. Those printers depend on the extruder end g-code not being written at the end of the print.

If you need the extruder end g-code at the global end g-code, just put this in the global end g-code:

    {machine_extruder_end_code, 0}

Or whichever extruder you need.

closed 2019-09-24 by Ghostkeeper

golfromeo-fr, 2019-09-24:
it is not because some do crap things that you should keep doing the same mistake

Ghostkeeper, 2019-09-25:
We're not going to drop support for dozens of printers because you don't like to type `{machine_extruder_end_code, 0}` in your custom end g-code! If you can find a way to do this while still maintaining support for printers that do have their complete end sequence in the end g-code, please let us know!

golfromeo-fr, 2019-09-27:
> We're not going to drop support for dozens of printers because you don't like to type `{machine_extruder_end_code, 0}` in your custom end g-code! If you can find a way to do this while still [...]

> {machine_extruder_end_code, 0}
we would need
{machine_extruder_end_code, current_active_extruder}
and the community have a place to write the explanation in a wiki 

it would solve properly the issue if you cannot solve ultimaker firmwares