Ultimaker/Cura #6399: Print Settings Hide Save Dialog

thesec, 2019-09-23:
Since release of v4 the print settings come at full height, hiding the - now separate - save dialog:
[image]

Please limit the settings' height to leave space for the dialog underneath as before so I do not have to close them prior being allowed to save.

smartavionics, 2019-09-23:
But you can drag the bottom of the settings by grabbing the three dots so you can alter the height of that panel.

fieldOfView, 2019-09-23:
Or you can install the Sidebar GUI plugin, which properly fixes this foolishness

thesec, 2019-09-23:
Hihi, excellent @smartavionics ... I just did not wear my jeweller's magnifiying lens to realize there being three fly sh... for dragging the panel in such a non-standardized fashion ;)

Yet I will leave this open. It should neither require a lens on the nose, nor user interaction or even a plugin to get around this.

rburema, 2019-09-24:
Also you can drag the panel around with the header.

rburema, 2019-09-24:
Shouldn't be completely full height though, it should stop (unless someone drags it) right above the 'action panel' (the one with the slice button in it).

Do you have multiple different sized monitors you work on? Was it always like this since first startup of 4.0.0?

thesec, 2019-09-24:
@rburema - it was always like it. Yes, I have two differently sized displays I work on.

thesec, 2019-09-24:
...
- a icon that is more obvious than the rotated hamburger (does it make it a hot dog?), see https://www.visualpharm.com/free-icons/mouse/drag%20mouse for commonly used ones
...

tablatronix, 2019-11-02:
This new UI has some annoyances, mostly that it makes all the UI elements on total opposite sides of the screen, everything is as far away from each other as possible.

The time slider and layer sliders should be next to each other, this pushes one all the way to the right, going back and forth is a pain. Controls are all the way on the left, viewport, all the way on the bottom, view control top left, layer control, the whole top, but the dialog is on the left ?

God forbid you have one of those widescreen monitors... lol

And an actual BUG
**The floating side bar covers the layer # popup..**

fieldOfView, 2019-11-02:
There is the Sidebar GUI plugin in the Marketplace, which solves most of these issues.

tablatronix, 2019-11-02:
@fieldOfView I know I installed it, it was awesome to have it back!, but I turned it off for a bit while I narrow down the source of my performance issues, not sure if it was related or not

fieldOfView, 2019-11-03:
OT: So, was it the cause of the slowdowns?

Ghostkeeper, 2019-11-04:
> And an actual BUG
The floating side bar covers the layer # popup..

Yeah, we found that as well and fixed it in Cura 4.4. See https://github.com/Ultimaker/Cura/pull/6499 and https://github.com/Ultimaker/Cura/pull/6592.

tablatronix, 2019-11-04:
So you fixed this?
[image]

Ghostkeeper, 2019-11-05:
No, as I said, for the floating settings panel we still don't have a real solution. Other than moving the panel a bit to the left, that is.

nallath, 2019-11-05:
> So you fixed this?

We have a solution for it in https://github.com/Ultimaker/Cura/pull/6607

rburema, 2019-11-07:
That's pulled into master though, so it's not in 4.4

closed 2024-12-07 by github-actions[bot]