Ultimaker/Cura #6401: wrong trajectory odd number of nozzle diameters

SnuSnurik, 2019-09-23:
if the part thickness is a multiple of an odd number of nozzle diameters, an incorrect trajectory is generated.

[circle.3mf.zip](https://github.com/Ultimaker/Cura/files/3642923/circle.3mf.zip)
[rectangle.3mf.zip](https://github.com/Ultimaker/Cura/files/3642924/rectangle.3mf.zip)
[image]
[image]

Ghostkeeper, 2019-09-24:
Since the ticket isn't very descriptive about what's wrong with it, here is a video:

http://dulek.net/work/wall_line_order.mkv

Ghostkeeper, 2019-09-24:
Ah, I see what's happening here.

Since the wall is EXACTLY 0.12mm wide and the line width is exactly 0.4mm, the centreline of the inner wall is technically an undefined shape, a ring with zero thickness. Due to rounding to floating point in the file format and then to micrometres in Cura, the polygon sometimes becomes 500 nanometres ticker and sometimes 500 nanometres thinner. This makes the polygon 0 thickness in some vertices.

The contour breaks up into multiple loops then, so it's running back and forth to draw the contours of each of the loops. However then Cura sees an extrusion move with 0 extrusion and turns that into a travel move that can take shortcuts.

Changing the inner wall line width to 0.399mm instead of 0.4mm fixes this.

SnuSnurik, 2019-09-25:
I've already changed the thickness of the inner wall to 0.399
but there remains one empty passage . It prints the inner wall, then one outer wall, then it tries to print the inner wall again, and then it prints the outer wall. 
I stumbled upon it on a more complex detail.
Why is a square with a wall thickness of 1.2 mm cura prints both the two outer walls and the filling, and the circumference (curve) tries to fit the inner wall and no parameter changes affect it.

p.s. here is an example of a model with a wall thickness of 1.2 mm (inner wall 0.399) estimated printing time 2 hours 40 minutes 16 grams of plastic.
and the second example of the model with a wall thickness of 1.6 mm estimated time was 2 hours 50 minutes and 20 grams of plastic
[1_2.3mf.zip](https://github.com/Ultimaker/Cura/files/3653568/1_2.3mf.zip)
[1_6.3mf.zip](https://github.com/Ultimaker/Cura/files/3653569/1_6.3mf.zip)

Ghostkeeper, 2020-11-19:
A fix for this issue is in progress here: Ultimaker/CuraEngine#1210.

Ghostkeeper, 2021-12-14:
This is fixed by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper