Ultimaker/Cura #6402: Changing infill to 100% set number of bottom layers to 999999

phazeman, 2019-09-23:
**Application version**
4.2.1

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Reproduction steps**
Set infill to 100% and watch number of bottom layers change to 999999

**Actual results**
 number of bottom layers changed to 999999

**Expected results**
Number of bottom layers should not be affected by infill percentage

**Additional information**

smartavionics, 2019-09-23:
That is how 100% infill is implemented. They chose to do it that way so it's not a bug but it could be a feature.

fieldOfView, 2019-09-23:
I believe the reasoning is that doing it this way you don't have walls and infill but just a layer that consists of walls, so there is no transition between walls and infill (in other words: more solidness).

phazeman, 2019-09-24:
> That is how 100% infill is implemented. They chose to do it that way so it's not a bug but it could be a feature.

> I believe the reasoning is that doing it this way you don't have walls and infill but just a layer that consists of walls, so there is no transition between walls and infill (in other words: more solidness).

It's definitely a bug since once you set the 100% infill - you are getting 0 top layers (the model is not being "closed" on top) and some other stuff...

fieldOfView, 2019-09-24:
You can set the infill to 99.999 or to 100.001 as a workaround.

Can you elaborate what you mean by the model not being closed? Bottom layers are solid, even if they are on top of the geometry.

fieldOfView, 2019-09-24:
> You can set the infill to 99.999 or to 100.001 as a workaround.

Or, a better workaround is to set the bottom walls back to the value you expected.

Ghostkeeper, 2019-09-24:
> It's definitely a bug since once you set the 100% infill - you are getting 0 top layers (the model is not being "closed" on top) and some other stuff...

The model should be closed on top as long as the model has fewer than 1 million layers. And since 1 million layers at the minimum layer height is still 1 metre, that should be fine.

ungutknut, 2019-10-11:
I still think this behaviour is undesirable since for top/bottom layers there aren't as many options as for 100% infill. Very useful stuff like "infill before walls" or "connect infill lines" is being ignored then. I guess no user would expect these features to suddenly stop working when setting infill to 100%.

Ghostkeeper, 2019-10-14:
The zigzag pattern is still available for skin, and the Connect Top/Bottom Polygons setting as well.

There is indeed no replacement for Infill Before Walls. That is a consequence of there not being a border between skin and infill any more. They HAVE to be printed at the same time then.

We've set this to be the default because for most people it would be easier to get strong parts this way. Is there a way in which we can make the behaviour more as expected for expert users but still strong by default for beginner users?

closed 2021-05-06 by Ghostkeeper