Ultimaker/Cura #6403: Hide buildvolume when opening custom stage

thopiekar, 2019-09-23:
I created a new stage and would like to undisplay the buildvolume. So only the loaded part is visible. Probably if I make the buildvolume invisible, it will also make my part invisible, I guess, but why is the following code making no difference?
I would expect that whenever I switch to my stage, it would make the buildvolume invisible. Even when I do `buildvolume = buildvolume._platform` (which should be the SceneNode of the buildplatform), I can't make it invisible, too.

```
    def onStageSelected(self):
        buildvolume = Application.getInstance().getBuildVolume()
        if buildvolume.isVisible():
            buildvolume.setVisible(False)

    def onStageDeselected(self):
        buildvolume = Application.getInstance().getBuildVolume()
        if self._is_buildvolume_hidden:
            buildvolume.setVisible(True)
```

Can you give me a hint how to make everything in the 3D view invisible except of the loaded model?

Thanks! :smile:

nallath, 2019-09-24:
Because we messed that bit up :$ 

The build volume (and the platform) have their own rendering code, which simply forgets to take visibility into account. 

I've fixed that in master now.

closed 2019-09-24 by nallath

rburema, 2019-09-24:
@nallath fixed it already: https://github.com/Ultimaker/Uranium/commit/f0c78e3e09ac3253bc514e81576706437dc2f5dd

thopiekar, 2019-09-24:
Great! Thank you all for the quick reply and direct support! :)