Ultimaker/Cura #6404: GUI: placing 3D - models in slicer; button for 90 degree rotation

zyxtaxbo, 2019-09-23:
When placing a model in the slicer, You can use dragger - controls to rotate the model around all three axis (default with 15 degree steps).
In most cases, You have to rotate the model by 90 degrees around certain axis.

There should be a dedicated button within the rotation dialog to rotate the model by +/-90 degrees around certain axis.
Or there should be a key combination (eg. hold "shift" while drag for 90 degree rotation and/or "alt" while drag for step-less rotation)

Should be an improvement for most users.

Ghostkeeper, 2019-09-24:
It's quite easy to hit the 90 degrees though by dragging with 15 degree increments, right? Or are you accidentally hitting 75 degrees or something?

zyxtaxbo, 2019-09-25:
Exactly, it could happen, that someone accidentally misses the 90 degree configuration and end up with 75 degree (or 105).
Dedicated buttons would make this process more reliable and definitely more convenient.

sui77, 2019-11-25:
Just wanted to propose this same feature, too. It would be so easy to just click a 90° button twice to turn the object upside down. I always somehow get stuck at 75° and don't know in which direction to further move the mouse then undoing and starting over from a slightly different perspective. Actually for X/Y rotations anything except 90° seem to be rare special cases to me.

loneFunction, 2020-04-01:
Yes!, or at least have a number entry filed like the move location fields.

For me it's easy enough to rotate to 90, but only if I grab the correct circle, which can be hard at certain views.
95% of the time select face and rotate 90 increments will be the only options needed.

Ghostkeeper, 2020-05-27:
Fixed here: https://github.com/Ultimaker/Uranium/pull/612

closed 2020-05-27 by Ghostkeeper