Ultimaker/Cura #6405: Postprocessing: Change on height/layer

tedukes, 2019-09-24:
I followed CHEP's video, Multi-Color 3D Print in CURA 3 on ENDER 3, https://www.youtube.com/watch?v=1nBnVtOEAiY&list=WL&index=2&t=0s

I'm using Cura 4.2.1, Ender 3 Pro, SKR 1.3, 2208 UART drivers, Marlin 2.0x

Tried multiple times, the steppers don't lock and any slight movement of the print head, axis, bed,  results in disaster.

Thought about trying the change filament script but I ruined a $15 PEI sheet trying the change at layer, so I haven't.

Why would there be a change filament script and change at height/layer script? Seems strange.

A good feature would be adding a beep at the pause and locking steppers on any pause.

Ghostkeeper, 2019-09-24:
The Filament Change script calls the `M600` filament change g-code. It is similar to the pause at height script but calls a different g-code. The intention is that you could need to do a manual operation during the pause, such as inserting a nut to lock it inside the print or something.

The Change at Layer script modifies the g-code such that certain properties of the print are (temporarily) changed, like the temperature. It works in a completely different way. It can also affect multiple layers and such.

We're assuming that the firmware can decide on whether the motors need to stay powered during a pause. Apparently your firmware disables motor power during a pause? We're not explicitly disabling power to the motors from the g-code or anything.

nyxaria, 2020-07-30:
Did you end up finding out what in Marlin made the motors go unpowered?

closed 2020-08-17 by no-response[bot]