Ultimaker/Cura #6407: [master] profile contains wrong settings

smartavionics, 2019-09-24:
**Application version**

master as of 24/09/2019

**Platform**

Linux

**Printer**

Kosell XL

**Reproduction steps**

Updated cura/uranium to latest master branch and opened cura.

**Actual results**

Some profile settings are wrong.

**Expected results**

Profile settings should have the same values as they had before the update.

**Additional information**

Look at this screen shot and see how the values displayed in the profiles dialog do not correspond to the values shown in the settings area on the right. Line width, infill line width, infill pattern, connect infill lines, etc. are all wrong.

[image]

Ghostkeeper, 2019-09-24:
We discovered this regression on the master branch as well. Essentially if you save a new profile it's not going to remember the settings in that profile, and if you update the profile it becomes corrupt completely.

smartavionics, 2019-09-24:
Thanks for the info. I have reverted back to recent usable master branches and will wait until this is fixed before merging from upstream again.

smartavionics, 2019-10-04:
I'm sure that you are aware that this regression still exists in the master branch(es). Do you intend to fix it?

fieldOfView, 2019-10-04:
I think they are planning to make the issue go away by replacing the code in the "intents" work that is going on in different branches.

smartavionics, 2019-10-04:
Thanks, I suspected that was the plan. Until either the regression gets fixed or the intents stuff is merged and usable I can't merge any upstream changes. No big deal.

Ghostkeeper, 2019-10-07:
Yeah the plan this week is to make the right hand side of that screen display the settings of the current profile with intents. The expected behaviour is (afaik):

* If the selected profile is an intent profile or a quality profile with default intent, show a combined view of both the intent profile and the quality profile.
* If the selected profile is a custom (quality changes) profile, show only the custom profile settings.