Ultimaker/Cura #6408: Additional Support settings

Asterchades, 2019-09-24:
**Is your feature request related to a problem? Please describe.**
I've recently been toying around with a few other slicers and have come back to Cura for the dimensional accuracy. While I was "away" I found a couple of useful features related to supports that I came to appreciate and I'd rather like to see implemented as I feel they make a potentially relevant difference for little to no performance impact in both calculation and printing.

**Describe the solution you'd like**
There are two features I came across that I'd like to see at least considered:

1) Alternating support directions. This basically involves rotating the selected support pattern by a specified amount on successive layers. For example, using the "Line" or "Zigzag" pattern, you could elect to rotate the pattern by 90° each layer, effectively creating a grid pattern with half the material usage.

The idea here is that more of the construct or interface is exposed to a closer amount of support material. In addition, having the support fill supported in two axes at once should increase the stability of the supports without using additional material At the same time, however, the supports are also rendered more brittle, thus easier to remove, without significantly compromising it.

This could also be applied to the Grid and Triangles patterns for similar effect. These would be at 45° and 60° angles respectively, or indeed have the whole lot able to be assigned by the user manually, either by specifying angles or number of layers per complete rotation. I cannot see it being of any real use for Gyroid or Cross patterns, and naturally it would do nothing for Concentric.

_This suggestion comes from a similar feature in IdeaMaker, where supports can be assigned a list of possible orientations to cycle through. I found it worked extremely well with the Line pattern alternating between 0° and 90°._

2) Offset support layering. This one is for situations where you're using thicker layers ("thicker" being as little as 0.2mm), but don't necessarily want to have a whole layer gap between your supports and your structure. The method that I've seen involves a unique first layer for the supports consisting of a platform (basically a single layer raft for the supports, though not necessarily at 100% density) of a thinner layer height.

For example, say you're using a 0.2mm layer height, but want a 0.12mm gap between the supports and the structure instead of 0.2mm. This "platform" layer would be printed first at just 0.08mm tall, and all following support layers would be offset by the same +0.8mm. Printing would then take place in the following order: 0.08mm support platform, 0.2mm model 0, 0.28mm support 0, 0.4mm model 1, 0.48 support 1, 0.60 model 2, and so forth.

As an added benefit (at least in my mind) this should also allow faster movement for both structure and support layers, since they're now offset from each other on Z. Effectively it is "Z-hopping" the counterpart on every layer. It would, however, require some kind of pre-determined interaction with the specified build plate adhesion method - rafts may have to go underneath it, while brims and skirts either need to be printed at the same reduced height or after the support surface.

Naturally this would double the number of Z moves on layers with supports present. I'm unsure if this would present an issue.

_This suggestion comes from similar functionality in Prusa Slicer (and possibly Slic3r), which appeared to do this automatically in my testing. It felt like a good idea and I don't really see a down side._

**Describe alternatives you've considered**
I'm honestly not really sure about alternatives to these, beyond perhaps shifting the "reduced' support layer from the bottom to the top (prior to any interface). This would reduce the number of additional Z steps required, while at the same time negating any potential gains (or losses) from printing the main structure and the supports vertically offset from each other.

**Affected users and/or printers**
Realistically this should be applicable to any and all printers without any modification to the printer's hardware or firmware, regardless of what movement method they use. Cartesian, be it Prusa style or Ender 5 (is there a name for that style?), CoreXY, Delta, Quadrap, Polar, Scara, or even Tripteron (including co-linear) should all be able to do it just fine.

**Additional context**

Ghostkeeper, 2019-09-24:
2) is a duplicate of this issue: https://github.com/Ultimaker/Cura/issues/6105

Reading through that again, there is a solution but it's technically very difficult to achieve for us.

closed 2019-09-24 by Ghostkeeper

Asterchades, 2019-09-24:
My apologies. Figured there was something I wasn't taking into account, and that certainly seems like it. Double-slicing is a lot of additional effort for what is ultimately not a massive gain.

Carry on! I look forward to testing 4.3 tomorrow after some sleep.

MariMakes, 2023-11-16:
Quick update on our side, you can now enter values that deviate from your layer height for the Support Z Distance. 
Can you please try it out and let us know if it resolves your issue?
You can download it here. https://github.com/Ultimaker/Cura/releases/tag/5.6.0-beta.1

Asterchades, 2023-11-16:
It certainly looks as though it should work. Though I will admit I had to double-check it a few times due to the nature of the preview screen - the narrower extrusions representing the thinner layer height had me second guessing whether it was suitable, but a manual inspection of the GCode (and a second opinion from another viewer) makes it clear that the shorter layer height is being respected.

This... could be an interesting one to tune - especially for miniatures.

NB It does not appear to be working with Tree-type supports - with or without an interface. Is this perhaps by design?