Ultimaker/Cura #6409: When missing profiles, include option to remove JUST the missing ones

AbeFM, 2019-09-24:
**Is your feature request related to a problem? Please describe.**
I constantly find myself with missing profiles, my options are manual clean up, remember to carefully click 'ignore', or remove everything and lose all my settings.

**Describe the solution you'd like**
A button that removes the offending profiles would let me keep the ones which work.

[image]

This is probably specific to people doing dev work, or who own multiple printers.

Ghostkeeper, 2019-09-25:
We've considered this before, but it can't really be done because the corruptions is 95% of the times caused by broken references between these files. Deleting these files then only causes more corruptions the next time you start Cura, because it's deleting files that other profiles and stacks are referencing to.

And we can't rely on the references themselves to discover which additional profiles should be deleted that are not mentioned in the message, because ostensibly these references are broken.

smartavionics, 2019-09-25:
I can normally recover from this situation by just deleting the files it mentions and the associated printers.

AbeFM, 2019-09-25:
Me too. Which is why I think it's something a computer could do. :-)

GK: Worst case, say I had to do it 2 or 3 times. It's still a lot better
than remembering DEAR GOD DON'T CLICK RESET several tens of times a week. I
agree it's no elegant, but it'd work. Similarly - you could automate that
by deleting files and immediately restarting CURA, deleting anything
troubling, and looping until there's no errors.

AbeFM, 2019-11-05:
> I can normally recover from this situation by just deleting the files it mentions and the associated printers.

A more thoughtful response is:
It seems in conflict with what GK mentioned in the previous post - his claim is that deleting the files causes more corruption. My experiences more parallel SA's.

If SA's right, then automated deleting makes sense. If GK's right, it doesn't.

Can we quantify how much of a problem just deleting anything that doesn't load would be? Or, moving to a "broken" sub-directory or similar?

Ghostkeeper, 2019-11-07:
We might be working with different types of workflows then. The errors I'm encountering are usually the result of fuzz testing with fairly complex combinations of profile operations, like duplicating materials that don't have a generic material, creating a custom profile, then switching printers, then editing the compatible material diameter, then switching back or something. Different people have different workflows and might only encounter the cases where simply deleting the broken profile is enough. For instance if you tend to manually edit the profiles in your config folder (which is something I very rarely need to do).

I have a bit of history about this issue and a bit of news as well:

1. Originally this message was added as a "last resort" error catching mechanism. This was necessary for a refactor we made for performance of profile switching, instead of crashing, in the interest of deadlines. Later we started using this as well as defensive coding.
2. In version 4.4 the original code that this message was made for has been refactored away. Some of the original idea is still there but the QualityManager code was very hard to maintain. However we expect that you'll see the errors much less often as long as you didn't inherit broken configuration from an older version. You'll still see it if you've manually edited your configuration.