Ultimaker/Cura #6410: Display travels in top-visible layer differently

drewnoakes, 2019-09-25:
It can be hard to see which travels belong to which layer.

Here is a screenshot that gives an example of what I mean:

[image]

Zooming in doesn't always help:

[image]

To that end, I would like the travels in the 'current' layer (by which I mean the top-visible one) to be distinct from those below it.

Two ideas come to mind:

- Show the top layer's travels using a different colour
- Allow hiding travels below the top layer

My goal is to better understand why this model will spend over half its time in travels, and see if there's anything I can do to reduce that time.

This should benefit anyone who is investigating issues with travels.

rburema, 2019-09-25:
You can already show (a) specific layer(s) if you drag the scroll-points on the right in preview mode.

[image]

drewnoakes, 2019-09-26:
@rburema thank you, that addresses my requirement perfectly and in a much more general way!

closed 2019-09-26 by drewnoakes