Ultimaker/Cura #6411: Cura 3.4.1 failed to load gcode - produced in the very same program

t-mou, 2019-09-25:
**Application version**
3.4.1

**Platform**
Windows

**Printer**
Cocoon model maker

**Reproduction steps**
creating gcode with cura, trying to open it afterwards

**Actual results**
failed to load

**Expected results**
loads

**Additional information**
Just setting up Cocoon printer and trying it out for the first time. I imported .stl file into cura, sliced it, saved the .gcode file and just wanted to open it again in cura before printing.
Parsing seems to not even start and it says Failed to load. Gcode file attached.
[print.zip](https://github.com/Ultimaker/Cura/files/3650438/print.zip)

rburema, 2019-09-25:
Does this reproduce in a recent version? Like 4.2.1 or 4.3.0?

Ghostkeeper, 2019-09-25:
Loading that g-code in current Cura results in the following error in the log:

```
2019-09-25 11:29:07,691 - ERROR - [Thread-4] UM.Logger.logException [88]: Traceback (most recent call last):
2019-09-25 11:29:07,691 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/FileHandler/ReadFileJob.py", line 66, in run
2019-09-25 11:29:07,692 - ERROR - [Thread-4] UM.Logger.logException [88]:     self.setResult(self._handler.readerRead(reader, self._filename))
2019-09-25 11:29:07,693 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshFileHandler.py", line 32, in readerRead
2019-09-25 11:29:07,693 - ERROR - [Thread-4] UM.Logger.logException [88]:     results = reader.read(file_name)
2019-09-25 11:29:07,693 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Uranium/UM/Mesh/MeshReader.py", line 19, in read
2019-09-25 11:29:07,694 - ERROR - [Thread-4] UM.Logger.logException [88]:     result = self._read(file_name)
2019-09-25 11:29:07,694 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/../plugins/GCodeReader/GCodeReader.py", line 74, in _read
2019-09-25 11:29:07,694 - ERROR - [Thread-4] UM.Logger.logException [88]:     node = self.readFromStream(file_data, file_name)
2019-09-25 11:29:07,695 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/../plugins/GCodeReader/GCodeReader.py", line 68, in readFromStream
2019-09-25 11:29:07,695 - ERROR - [Thread-4] UM.Logger.logException [88]:     return self._flavor_reader.processGCodeStream(stream, filename)
2019-09-25 11:29:07,696 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/../plugins/GCodeReader/FlavorParser.py", line 421, in processGCodeStream
2019-09-25 11:29:07,696 - ERROR - [Thread-4] UM.Logger.logException [88]:     current_position = self.processGCode(G, line, current_position, current_path)
2019-09-25 11:29:07,696 - ERROR - [Thread-4] UM.Logger.logException [88]:   File "/home/ruben/Projects/Cura/cura/../plugins/GCodeReader/FlavorParser.py", line 268, in processGCode
2019-09-25 11:29:07,697 - ERROR - [Thread-4] UM.Logger.logException [88]:     f = float(item[1:]) / 60
2019-09-25 11:29:07,697 - ERROR - [Thread-4] UM.Logger.logException [88]: ValueError: could not convert string to float: '{TRAVEL_SPEED}'
```

This behaviour is correct because the g-code contains these strings `{travel_speed}` in its start-up and ending sequences, which is not valid g-code.

It seems that both your start and your end g-code contain the word `{travel_speed}` with the intention that this gets replaced with the travel speed setting's value. However it doesn't do that replacement since the setting name is `speed_travel`.

This is something that has been fixed in later releases. We now accept `travel_speed` as well as a replacement key, to improve support for people who copy-pasted their start and end g-code from Cura 15.04 which had different setting names. So please upgrade to a newer version where we fixed this bug.

closed 2019-09-25 by Ghostkeeper

t-mou, 2019-09-25:
yes, later version solved this. thank you.