Ultimaker/Cura #6413: After upgrading from 4.0 to 4.3 version lost Ultimaker materials information

rkorczynski, 2019-09-25:
I lost Ultimaker materials information after upgrading from 4.0 to 4.3 version.

Ghostkeeper, 2019-09-25:
What do you mean with "lost"? The materials are no longer in the list of available choices?

rkorczynski, 2019-09-25:
That's exactly what I mean. The Ultimaker materials were on the available choice list in version 4.0 and after upgrading they disappeared. No more Ultimaker materials are on the list of materials to choose from.

fieldOfView, 2019-09-25:
What printer do you use?

rkorczynski, 2019-09-25:
Monoprice Maker Ultimate 3D.

Ghostkeeper, 2019-09-26:
We don't have a printer definition for Monoprice Maker Ultimate 3D. What printer did you add in Cura? Or did you install a mod for Cura to add it?

Ghostkeeper, 2019-09-26:
I'd say though, that according to the [specs](https://www.monoprice.com/product?p_id=15710) it's correct that the Ultimaker filaments are not shown for your printer since your printer accepts only 1.75mm diameter filaments, while all Ultimaker filaments are 2.85mm.

closed 2019-09-26 by Ghostkeeper

rkorczynski, 2019-09-26:
I installed as Monoprice Ultimate printer that I found on the list of available printers.

Ghostkeeper, 2019-09-27:
The Monoprice Ultimate is using the [Wanhao D6 extruder](https://github.com/Ultimaker/Cura/blob/648f167911134ce15ac7b9e44619bfa47589817f/resources/definitions/monoprice_ultimate.def.json#L15) according to our profiles. This extruder has been using [1.75mm](https://github.com/Ultimaker/Cura/blob/648f167911134ce15ac7b9e44619bfa47589817f/resources/extruders/wanhao_d6_extruder_0.def.json#L14) filament for the past year or so. I wouldn't expect anything has changed with that recently.

rkorczynski, 2019-09-27:
Thank you for the info.