Ultimaker/Cura #6417: negative values/percentages have no "-" sign

Stormwing01, 2019-09-25:
**Application version**
4.3

**Platform**
Windows 7

**Printer**
Wanhao i3 mini

**Reproduction steps**
I just changed the infill overlap percentage to a negative value (to -10 percent in my case)
This happens for any setting, not just infill overlap percentage.

**Actual results**
The negative value doesn't show. it shows " 10" instead of "-10" but in the case of a negative value, there is a space before the "10" instead of a "-" as there should be

**Expected results**
It should show the "-" sign for negative values, otherwise it's easy to mistake it for a positive percentage.

**Additional information**

[image]

I hope the screenshot helps. it's... a small bug, but it's been there ever since  version 4.1 and I have included examples of how an older version of cure (4.0) does not have the bug, while version 4.3 does in the screenshot.

fieldOfView, 2019-09-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/6277, though I like your pictures and analysis.
We can see that between 4.0 and 4.3, the font rendering is somehow quite different, though it should be using the same font.

Ghostkeeper, 2019-09-26:
> We can see that between 4.0 and 4.3, the font rendering is somehow quite different, though it should be using the same font.

Actually we found that the Noto Sans font was apparently not being used on Windows. So maybe that's the difference then.

Ghostkeeper, 2019-09-26:
I'll close this ticket as duplicate of #6277 (and improve the title of that ticket a bit).

closed 2019-09-26 by Ghostkeeper