Ultimaker/Cura #6418: Cura 4.3 machine settings weirdness

ungutknut, 2019-09-25:
I just upgraded to Cura 4.3 (Win10, 64bit) from 3.6 (which I have uninstalled before upgrading) and have some troubles setting up a custom machine.

The values "X min" and "Y min" in the printhead settings dialog can't be changed. I'd like to set them to 42 and 32mm, but it doesn't accept that and changes back to default -20 and -10.

Second issue is the compatible material diameter which I'd like to change to 1.75mm. There seems to be some confusion about wheather to use "." or "," as a comma. If I use a dot, it will be ignored. If I use a comma, it's accepted but changed for a dot afterwards.

BTW: Eventhough I've uninstalled old Cura versions, Cura 4.3 still found settings for an old custom machine which were used by default. I didn't use that machine, but maybe it's got something to do with these issues?

Ghostkeeper, 2019-09-26:
The first one is expected behaviour. The shape of the print head is coordinates relative to your nozzle position (before the nozzle offset). If the minimum X position is positive, the maximum X position is also positive. That means that the entire print head must be in the positive coordinates. In other words, the nozzle is not inside the print head. This makes no sense, so we don't allow that to prevent you from making mistakes.

The second bug report I'm not able to reproduce:
* I changed the compatible material diameter to `1,75`. This accepts it but changes it to `1.75`. This is expected to be consistent with the other settings.
* After this change, the list of materials is the materials with 1.75mm filament. So like the Chromatik, Innofill, etc. and not the Ultimaker materials.

ungutknut, 2019-09-26:
> This makes no sense, so we don't allow that to prevent you from making mistakes.
Ok, thanks for clarifying. So the manufacturer of my printer has wrong values in it's own flavour of cura (which I don't use) and I just copied them. 

> The second bug report I'm not able to reproduce:

I tried to reproduce it on 2 PCs (one Win10, the other Xubuntu - both with Cura 4.3) and I can observe different behaviour than before:

Win10: It doesn't accept a dot like before (typing "1.75" is just "175"), but when I enter "1,75", the comma stays and isn't changed for a dot like before. Still this is strange as the nozzle size is separated by a dot.

Xubuntu: The dot is ignored like on Win10 (typing "1.75" is just "175"), but when I enter "1,75", it's auto-changed for "1.75".

I don't know if that even is a real problem, but it's at least an inconsistency. And there's a chance, people could unintentionally set the diameter to a massive 175mm. Here's a screenshot of the dialogue, just to make sure we're talking about the same thing:
https://imgur.com/p3rUBc0

Perhaps It's got to do something with my keyboard input layout/language, which is German (we don't use dot as comma separator).

Attached there's a project file with random settings and geometry.
[CFFFP_Winkel5.zip](https://github.com/Ultimaker/Cura/files/3657412/CFFFP_Winkel5.zip)

Ghostkeeper, 2019-09-26:
> Ok, thanks for clarifying. So the manufacturer of my printer has wrong values in it's own flavour of cura (which I don't use) and I just copied them.

Ah, I think that this was just outdated then and you should put negative signs before what the manufacturer provided you. We've recently changed the sign of this field around for clarity.

ungutknut, 2019-10-11:
Thx for the explanation. I close it now since the main issue is resolved and that "."/"," thing might not be very important.

closed 2019-10-11 by ungutknut