Ultimaker/Cura #6422: Question: Removing boundary box and graph paper

thopiekar, 2019-09-25:
As a little follow up on the question of how to replace, the next question:
Is it possible to remove the boundary box and graph paper, too?

Right now I have the following result: (Which is already very pleasant! - thank you again @nallath )
[image]
[image]

I couldn't any nice function that could give me the boundary box and the grid paper as done for the build platform. I guess that nobody asked for this ever before.

Could you give me a hint where I can find the critical code at least?
That would be awesome! 🤓

fieldOfView, 2019-09-26:
How is this a bug?

nallath, 2019-09-26:
I forgot to commit the fix I made that also enables the hiding of the build volume if you set it to invisible.

> How is this a bug?

If you set the visibility of the BuildVolume scene node, you'd expect it to be ya know. Invisible.

closed 2019-09-26 by nallath

thopiekar, 2019-09-26:
Great! Thank you again @nallath