Ultimaker/Cura #6424: Cura 4.3 - Strange behavior at start of print compared to 4.2.1

MetalRobo, 2019-09-26:
**Application version**
4.3

**Platform**
Windows 10

**Printer**
Anycubic Mega-S/i3 Mega

**Reproduction steps**
Slice anything

**Actual results**
Nozzle goes to home position, touches the bed, goes up as normal to clear any oozing filament, but then for some reason goes back down to the bed into the oozed filament then drags the ball of oozed filament across the bed to the starting point of the print, thereby ruining the print in most cases

**Expected results**
In Cura 4.2.1 the nozzle would go to the home position, touch the print bed, go back up a bit, then from the raised height would go directly to the start of the print, which allowed the oozed filament to stay in the corner of the print bed. This worked nicely and there was no reason to change it.

**Additional information**

The gcode that seems to have been added which causes the problem is attached and highlighted in green. On the left is Cura 4.2.1 and on the right is the same print with Cura 4.3
[image]

Psychonno, 2019-09-26:
Had the same happening (same printer). For now I went with deleting the second g-code line that zeros the printhead again in the machine settings. Hope that works, have yet to test...

rburema, 2019-09-26:
So if I understand correctly, _before_ the gcode that is shown, the nozzle would at a position _higher_ than the z 0.2/0.275 that is shown? (Otherwise it doesn't match with the situation you're describing.)

(It's not entirely true that there was no reason to change it, see [this issue](https://github.com/Ultimaker/Cura/issues/6172) and [the relevant fix](https://github.com/Ultimaker/CuraEngine/pull/1112) ... but that doesn't mean it can't be a bug of course!)

MetalRobo, 2019-09-26:
> So if I understand correctly, _before_ the gcode that is shown, the nozzle would at a position _higher_ than the z 0.2/0.275 that is shown? (Otherwise it doesn't match with the situation you're describing.)
 
That is correct. Here is the code before that point which shows Z15.0:
;FLAVOR:Marlin`
;TIME:2978
;Filament used: 2.06928m
;Layer height: 0.2
;MINX:92.494
;MINY:51.33
;MINZ:0.2
;MAXX:124.182
;MAXY:158.554
;MAXZ:69.8
;Generated with Cura_SteamEngine 4.3.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F100 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F100
M117 Printing...
G5
		
G92 E0
G92 E0
G1 F2400 E-6
;LAYER_COUNT:349
;LAYER:0
M106 S255
M204 S3000
M205 X10 Y10
G1 F600 Z0.275
G0 F3000 X99.925 Y55.336 Z0.275
M204 S1800
M205 X8 Y8
;TYPE:SKIRT
G1 F600 Z0.2
G1 F2400 E0`

Ghostkeeper, 2019-09-26:
This changed in 4.3 because we changed the layer switching code to ensure that the nozzle is at the height of the new layer before moving to the position that it needs to start extruding at. Previously it made the travel moves to the starting point of the new layer and then changed the Z. Now it first changes the Z and then moves to the starting point. The idea was that moving up to the higher layer would prevent hitting the walls of the previous layer so you'd have less scarring at no additional cost.

Because it now guarantees that it's at the height of the new layer before making the travel move to the start of the first line, it's also doing that for your case where the height of the new layer is lower than the current position. In that regard it's intended behaviour, though it looks a bit strange for your printer's start g-code. Is it really a problem though?

Psychonno, 2019-09-26:
Yes it is! Since the movement into the extruded material drags everything over the buildplate. It could be remedied with an additional short movement before lowering the toolhead to print height.

Ghostkeeper, 2019-09-26:
For Ultimaker's printers, the firmware dips the nozzle against the build plate to make sure that the primed material sticks to the buildplate and doesn't get dragged along with the nozzle towards your print. That could be worth experimenting with, maybe?

MetalRobo, 2019-09-26:
I'll take a video of the problem happening after work today and send it to
you to make the problem more clear

MetalRobo, 2019-09-26:
[Here is the video](https://streamable.com/qgx7y). Everything is good until about 0:18, where instead of starting the print like it did in 4.2.1, it lowers back on the filament that just came out and makes a big blob

Psychonno, 2019-09-26:
Yes, exactly like mine. I added a short movement (2cm) in Y-direction just before printing, it is enough to get the printhead clear of the extruded filament before it dips down.

Ghostkeeper, 2019-09-27:
The Ultimaker 3 also has a 20mm movement to the side like you're describing. Here is a video: http://dulek.net/work/um3_prime_blob.mp4

Maybe you'd like to put this in the start g-code?

MetalRobo, 2019-09-27:
Ok.... So... How would I go about doing this?

Psychonno, 2019-09-27:
Just go into settings - printer - machine settings (I think) and add a line directly before the print command in the start g-code. I used “G1 Y20 F2400“. Could be faster or further, but certainly resolves the  problem.

Ghostkeeper, 2019-09-30:
We could put that in the default profiles if you feel confident that this works.

MetalRobo, 2019-10-02:
I tested this out myself putting it directly into the gcode before printing
starts and here is the result.
https://streamable.com/azozx
So much better. Thanks Psychonno. So yes please put it into the default
profiles

Ghostkeeper, 2019-10-02:
I've added that g-code but using the travel speed setting rather than F2400. Seems like that is about what you intended. See my commit here: https://github.com/Ultimaker/Cura/commit/2a4c387067bbd9c25190cbe557a96bb391687048

I'll consider this resolved then. This is for us a much nicer solution than having to make exception cases for when the start g-code leaves the printer in various states. Thanks for the experimenting and testing!

closed 2019-10-02 by Ghostkeeper

Polgr89, 2019-11-15:
Hi all! 

I am pretty new about how to program  Gcodes for 3D print.
I think I have the same problem that you had, or at least very similar.

When I start printing, it seems as if everything goes fine, the nozzle is correctly calibrated and stays 0.2 mm from the bed. Then, when it starts printing goes up like 1 cm and prints on the air.
Do you know how I can solve this issue?
Hoping someone can help me.

Thank you all.

I copy my current Gcodes: 

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder

G91 ;Relative positionning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

Ghostkeeper, 2019-12-03:
I'd expect that g-code to leave your nozzle 1cm above the build plate, Polgr, but immediately afterwards it should go to the initial layer height. I think your problem is a little bit lower in the g-code. Could you show some g-code that was generated by Cura, up until it starts printing the skirt or brim?

Psychonno, 2019-12-03:
Do you make any postscript changes? I once had my printhead start in thin air after I miscalculated a "change x at layer height" command.

Smithy101, 2019-12-28:
I've been having an issue with this new start movement as well. 
The code posted by Metalrobo is exactly what I've been having to deal with. 
The problem here is that if you have a clip holding down your glass, the nozzle smacks into it. Or if your bed level is off by a hair on the start corner, same issue. 

I've been getting around this by deleting the extra G92 E0 line (Not sure why it's there twice) and a few lines down deleting the line G1 F600 Z0.275 in his example. 
This keeps the nozzle raised, then comes down to the print area and avoids knocking the nozzle into anything on the edge of the glass or the glass itself if your level is slightly off. (Yes, the level would have to be very off, but if you JUST made a change such as a new nozzle and running a test print, this can damage your printer if something isn't set right)

I understand why this change was made, but it's creating some issues where I now need to edit my gcode to remove those lines that lower the head back down prior to moving to print area. 
Can this new start be turned off  in the software?

Example code

;FLAVOR:Marlin
;TIME:256
;Filament used: 0.0897814m
;Layer height: 0.15
;MINX:186.709
;MINY:180.602
;MINZ:0.3
;MAXX:213.292
;MAXY:219.398
;MAXZ:5.55
;Generated with Cura_SteamEngine 4.4.1
M140 S70
M105
M190 S70
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
M107 ;Start with the fan off
G21 ;Set units to millimeters
G91 ;Change to relative positioning mode for retract filament and nozzle lifting
G1 F200 E-3 ;Retract 3mm filament for a clean start
G92 E0 ;Zero the extruded length
G1 F1000 Z5 ;Lift the nozzle 5mm before homing axes
G90 ;Absolute positioning
M82 ;Set extruder to absolute mode too
G28 X0 Y0 ;First move X/Y to min endstops
G28 Z0 ;Then move Z to min endstops
G1 F1000 Z15 ;After homing lift the nozzle 15mm before start printing

G92 E0
**G92 E0   <------- Duplicate**
G1 F3600 E-7
;LAYER_COUNT:36
;LAYER:0
M106 S255
**G1 F600 Z0.375    <----- Delete this line**
G0 F2400 X189.914 Y184.093 Z0.375
;TYPE:SKIRT
G1 F600 Z0.3

Dswal, 2019-12-30:
> I've been having an issue with this new start movement as well.
> The code posted by Metalrobo is exactly what I've been having to deal with.
> The problem here is that if you have a clip holding down [...]

Thank you! I am having the same issue since I upgraded Cura from 4.1 today. Trying to find out why I was getting an unusual line at the start of the print.

vice1981, 2020-09-23:
Hello it seems that i have a similar problem with cura. i think that the problem is the start gcode.
When i load a test gcode from teaching tech 3d i works fine, the owl test works fine but when i load something sliced in cura it just wont do a first layer print, instead it just go where it is supposed to go but without putting material or at least very little material witch result in sticking to the nosle.

Ghostkeeper, 2020-09-25:
@vice1981 that g-code instructs the printer to go to a corner, try extruding a bit of material without knowing where the filament is in the nozzle, and then move a little bit aside, before actually starting with the print. I think the behaviour you describe is expected then.
If you see anything wrong, we'll need to know what about the output g-code is wrong. We can't print with your printer so it's impossible for us to guess what we'd need to change to get your desired outcome.