Ultimaker/Cura #6425: Top Surface Skin Layers under counted when ironing is on

AbeFM, 2019-09-26:
**Application version**
~Master branch as of Sept 9th or so? (SA's build, can get details if pertinent)

**Platform**
win10

**Printer**
dual extrusion fff with one nozzle 'disabled'

**Reproduction steps**
Select 1 "Top Surface Skin Layer"
Select Ironing
Use different pattern for Top Surface Skin Layer than ironing and Top/Bottom Pattern

**Actual results**
Uppermost ironing layer is correct, but underlying topmost *printed* layer is using setting for "Top Bottom" pattern instead.

**Expected results**
Uppermost N layers selected as "Top Surface Skin layers" to be printed in Top Surface Skin Layers Pattern".

**Additional information**
If this isn't enough info, let me know!

rburema, 2019-09-26:
[image]

(See image) I see nothing wrong? Keep in mind that ironing is added to the top-layer, so in the preview it shows as one layer, so if you go one layer down it doesn't show 'the topmost layer without ironing' but instead the layer below that.

AbeFM, 2019-09-26:
Agreed. Sorry for the false alarm.

closed 2019-09-26 by AbeFM