Ultimaker/Cura #6430: [4.3] Overhanging wall angle changes feedrate if used with support blockers as modifiers

alfrix, 2019-09-27:
Using support blockers to modify settings that overlap with other models changes the feedrate, (even without selecting anything to modify) if "overhanging wall angle is set to less than 90"
Normal slice
[image]
with a support blocker set on "Modify settings that overlap with other models " and "overhanging wall angle 50"
[image]

nallath, 2019-09-27:
Could you share the project file of this? I've tried reproducing this with the following steps:
1. Add a model & print it with defaults
2. Add another model and use that as "modify settings for overlaps"
3. Set the support overhang angle of the modifier mesh to 90
4. Check the preview

I don't see any feed rate differences when doing that.

Ghostkeeper, 2019-09-27:
Yeah, this is easy to reproduce. It seems that the cutting mesh messes with the overhang detection. Here is a project file in a zip that reproduces the issue.

[overhanging_wall_detection.zip](https://github.com/Ultimaker/Cura/files/3662425/overhanging_wall_detection.zip)

alfrix, 2019-09-27:
> 3.Set the support overhang angle of the modifier mesh to 90

90 is disabled, it should be less than 90 (i used 50 for the example)
it happens on all the models i've tried

normal slice
[image]
overhanging wall angle 50 with cube set as modify
[image]

alfrix, 2019-09-27:
it's a lot more broken than i thought, maybe this behaviour is related, with a normal slice with no extra changes and a cube set as modifier (the example it's a benchy)

[image]

Ghostkeeper, 2019-09-30:
That is expected behaviour for a cutting mesh. It breaks up the model into several pieces and adjusts settings for each of them. CuraEngine is at this moment unable to change the line configs halfway through a contour, so they are made into separate contours.

Ghostkeeper, 2019-10-01:
The issue with the overhanging wall angle has been deferred to a day when we have blue skies. There are more important things for us to fix, especially since the Overhanging Wall Angle is still experimental.

alfrix, 2019-10-01:
Add bridge settings to the list, for the time being you can sort of workaround these problems, by disabling both the overhanging and bridge settings in the modifier mesh, and because of the contour limitation, the mesh should completely enclose or cut the entire object transversally as i show here:

[image]

not perfect, but still useful for some testing and advanced stuff

closed 2023-06-24 by github-actions[bot]