Ultimaker/Cura #6433: Ultimaker Cura 4.3.0 Printer Selection not available

tronicgr, 2019-09-27:
**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

tronicgr, 2019-09-27:
I have two printers, an Ultimaker3 and Ultimaker3 extended. They are setup as group with one of them being the host using the Cura Connect.
 
In previous Cura version, when I pressed the button to "Print over network" it would popup a printer selection menu to choose which one to send the print to. Now after installing the Cura 4.3.0 this window does not pop up anymore and the print is send to the first available. This is inconvenient as I have different color material on each and I'd like to chose which one can print...

I have Windows 10 Enterprise.
 
Can you bring back that printer selection menu? Is there a setting somewhere to enable it back?

Ghostkeeper, 2019-09-30:
I'm told it was intentionally removed. It was no longer possible due to a refactor that was made during this release. I'll ask the developers of that plug-in on whether it needs to be put back.

tlum2, 2019-10-03:
Not being able to select from our 4 printers is a huge feature deletion, basically it is now taking us back to putting everything on thumb drives so we can select the correct printer. Please consider reinstating this feature.

ChrisTerBeke, 2019-10-25:
Addressed by https://github.com/Ultimaker/Cura/pull/6580.

closed 2019-10-25 by ChrisTerBeke