Ultimaker/Cura #6435: Unable to set Z hop speed above 3mm/s

Haxk20, 2019-09-27:
**Application version**
4.2+

**Platform**
Arch Linux

**Printer**
Tevo Tarantula profile (Printer is really TT Pro)

**Reproduction steps**
Enable Z hop and set speed above 3mm/s

**Actual results**
Cura complains about the speed abou 3mm/s and wont slice the model

**Expected results**
Slice the model properly with the settings applied

Ghostkeeper, 2019-09-30:
That is because the Tevo Tarantula printer definition specifies that the firmware limits the Z speed to 3mm/s:

https://github.com/Ultimaker/Cura/blob/05a941875582a748b55a7ab4e8d8e290b0961115/resources/definitions/tevo_tarantula.def.json#L44

From our perspective this is expected behaviour. We disallow setting speeds to higher than what your firmware accepts.

Ghostkeeper, 2019-09-30:
If you can show that the firmware accepts greater speeds, we can change it for you if you like.

closed 2019-09-30 by Ghostkeeper

Haxk20, 2019-09-30:
Ahh i see. But im not talking about original Tevo Tarantula. I have the new
TEVO tarantula PRO. That one can surely move at much higher speed then
3mm/s because when i start a print and nozzle moves up its so fast that i
was worried about it. But indeed it is how its supposed to be in firmware
side so if it would be possible to maybe create a new profile for TEVO
tarantula PRO that would have Z speed higher. @ghostkeeper

Ghostkeeper, 2019-09-30:
It is possible but it's hard for me to help with that since I don't have your profile.

One thing you can do is to go into the Marketplace and find the "Machine Settings" plug-in by fieldOfView. It allows you to change the "Maximum Speed Z" setting. Increase that and then you should be allowed to set a greater Z hop speed.

Haxk20, 2019-09-30:
Ahhh i see. Thank you.