Ultimaker/Cura #6436: Geschwindigkeit für überhängende Wände

lwbsbde, 2019-09-27:
Cura 4.2.1

Hallo
Die Geschwindigkeit der überhängenden Wände wird beim Druck mit Support nicht angepasst. 
Oder es fehlt die Einstellung für die Geschwindigkeit bei Druck auf Support.
 Besonders bei Verwendung von Dualextruder und PVA .

Ghostkeeper, 2019-09-30:
This is intended. The overhanging wall speed only applies to overhang that is not supported. There is no setting right now to adjust the speed of the layer printed on top of support, only the fan speed (Fan Speed Override).

I'll clarify this in the setting description.

closed 2019-09-30 by Ghostkeeper

lwbsbde, 2019-10-01:
How can I print slow on PVA support

Ghostkeeper, 2019-10-02:
We don't have a specific set of settings for the layer above support, other than the fan speed via the Fan Speed Override setting. You could achieve that with a separate mesh but it's going to be very hairy to model that. So it's honest to say then that Cura doesn't support that at the moment.

lwbsbde, 2019-10-03:
I also miss a customizable Fan speed when overhanging Wall without support. Experimental Settings.

Overhanging Wall Angle
Overhanging Wall Speed
+Overhanging Wall Fan Speed

commit 5c2c4118 (2019-09-30): Add clarification about that supported overhang doesn't fall into this category

Fixes #6436.