Ultimaker/Cura #6443: Post Processing script settings not saved in project 4.2.1->4.3.0

photodude, 2019-09-28:
Cura 4.3.0
project from Cura 4.2.1

System
Windows 10

Printer
BIBO 2

steps
Saved a project saved in Cura 4.2.1 with post processing scripts
Opened a project with Cura 4.3.0 expecting to see the post processing scripts "ChangeAtZ5.1.1"

**Actual results**
no post processing scripts were in the project.

**Expected results**
post processing scripts  "ChangeAtZ5.1.1" would be there when the project was opened

Needed to make changes to an old project Temperature tower, the  "ChangeAtZ5.1.1" post processing sets up the temperatures changes for the layer to properly conduct the test.

Ghostkeeper, 2019-09-30:
I'm not able to reproduce this bug. I've saved the project file in 4.2 and it produced this file: [test.3mf.zip](https://github.com/Ultimaker/Cura/files/3670837/test.3mf.zip)

This project file contains the post-processing script, as can be seen when opening the .3mf file with a zip archiver application. In the file `/Cura/BIBO2 dual.global.cfg` it lists the post-processing script and its settings.

When opening this file again in Cura 4.3 it loads the post-processing script fine. When slicing it replaced all of my `;LAYER` comments with the common typo `;LYAER`.

Maybe you can provide the project file that you had that reproduced the problem? Then we can see whether it was 4.2 that saved it wrong or 4.3 that loaded it wrong.

photodude, 2019-09-30:
@Ghostkeeper I'll see about reinstalling Cura 4.2 and recreating. I saved over the current file from 4.3 (I have not reopened it to see if it has the layer change at z saved)

[BSE1_TempTower_v2.zip](https://github.com/Ultimaker/Cura/files/3672032/BSE1_TempTower_v2.zip)

Ghostkeeper, 2019-10-01:
That 3MF file contains the data for 4 post-processing scripts:

```
post_processing_scripts = [ChangeAtZ]\\\na_trigger = layer_no\\\nb_targetZ = 5.0\\\nb_targetL = 117\\\nc_behavior = keep_value\\\nd_twLayers = 1\\\ne1_Change_speed = False\\\ne2_speed = 100\\\nf1_Change_printspeed = False\\\nf2_printspeed = 100\\\ng1_Change_flowrate = False\\\ng2_flowrate = 100\\\ng3_Change_flowrateOne = False\\\ng4_flowrateOne = 100\\\ng5_Change_flowrateTwo = False\\\ng6_flowrateTwo = 100\\\nh1_Change_bedTemp = False\\\nh2_bedTemp = 60\\\ni1_Change_extruderOne = True\\\ni2_extruderOne = 194\\\ni3_Change_extruderTwo = False\\\ni4_extruderTwo = 190\\\nj1_Change_fanSpeed = False\\\nj2_fanSpeed = 255\\\n\\\n
	[ChangeAtZ]\\\na_trigger = layer_no\\\nb_targetZ = 5.0\\\nb_targetL = 226\\\nc_behavior = keep_value\\\nd_twLayers = 1\\\ne1_Change_speed = False\\\ne2_speed = 100\\\nf1_Change_printspeed = False\\\nf2_printspeed = 100\\\ng1_Change_flowrate = False\\\ng2_flowrate = 100\\\ng3_Change_flowrateOne = False\\\ng4_flowrateOne = 100\\\ng5_Change_flowrateTwo = False\\\ng6_flowrateTwo = 100\\\nh1_Change_bedTemp = False\\\nh2_bedTemp = 60\\\ni1_Change_extruderOne = True\\\ni2_extruderOne = 203\\\ni3_Change_extruderTwo = False\\\ni4_extruderTwo = 190\\\nj1_Change_fanSpeed = False\\\nj2_fanSpeed = 255\\\n\\\n
	[ChangeAtZ]\\\na_trigger = layer_no\\\nb_targetZ = 5.0\\\nb_targetL = 335\\\nc_behavior = keep_value\\\nd_twLayers = 1\\\ne1_Change_speed = False\\\ne2_speed = 100\\\nf1_Change_printspeed = False\\\nf2_printspeed = 100\\\ng1_Change_flowrate = False\\\ng2_flowrate = 100\\\ng3_Change_flowrateOne = False\\\ng4_flowrateOne = 100\\\ng5_Change_flowrateTwo = False\\\ng6_flowrateTwo = 100\\\nh1_Change_bedTemp = False\\\nh2_bedTemp = 60\\\ni1_Change_extruderOne = True\\\ni2_extruderOne = 212\\\ni3_Change_extruderTwo = False\\\ni4_extruderTwo = 190\\\nj1_Change_fanSpeed = False\\\nj2_fanSpeed = 255\\\n\\\n
	[ChangeAtZ]\\\na_trigger = layer_no\\\nb_targetZ = 5.0\\\nb_targetL = 445\\\nc_behavior = keep_value\\\nd_twLayers = 1\\\ne1_Change_speed = False\\\ne2_speed = 100\\\nf1_Change_printspeed = False\\\nf2_printspeed = 100\\\ng1_Change_flowrate = False\\\ng2_flowrate = 100\\\ng3_Change_flowrateOne = False\\\ng4_flowrateOne = 100\\\ng5_Change_flowrateTwo = False\\\ng6_flowrateTwo = 100\\\nh1_Change_bedTemp = False\\\nh2_bedTemp = 60\\\ni1_Change_extruderOne = True\\\ni2_extruderOne = 220\\\ni3_Change_extruderTwo = False\\\ni4_extruderTwo = 190\\\nj1_Change_fanSpeed = False\\\nj2_fanSpeed = 255\\\n\\\n
```

I can't open that file with Cura though because it seems to use a custom definition file BIBO single E1 that I don't have.

photodude, 2019-10-01:
@Ghostkeeper excellent. Good to know 4.3 is saving correctly in the file. I'll see about trying to reproduce with a file saved from 4.2
here is a link to BIBO's provided settings (their definitions need some work, fortunately they are making the changes as I've requested them) https://drive.google.com/file/d/0B76TjeOkslHlZ2VBLXBSdzJYdms/view?fbclid=IwAR0HLKR3LPBmX7q-O8jeg2CGPMIaihWV_RLCiAfA4QV3shIzjzpAqFIAXcI

Ghostkeeper, 2019-10-02:
With those BIBO definitions loaded, I can open the 3MF file as a project file, but then it also properly displays the post-processing scripts. Seems like it loads fine! I'm confused :confused:
[image]

photodude, 2019-10-02:
@Ghostkeeper I'll try to recreate the issue with 4.2 it will be a couple of days hopefully before the weekend.

photodude, 2019-11-25:
As this is working in 4.3 and I don't think I'll find time to test 4.2.1 again with 4.4 now released; I'm considering this issue as unreproducible and closing.

closed 2019-11-25 by photodude