Ultimaker/Cura #6444: [4.2.1] Preview of initial layer not representative of actual nozzle width

cmgreyhounds, 2019-09-29:
**Application version**
4.2.1

**Platform**
Windows 8.1

**Printer**
Ender 3

**Reproduction steps**
Using a 0.4 nozzle with 0.4 line width and setting the Initial Layer Line Width to 150%.

**Actual results**
The Cura graphical Preview depicts the first layer as being essentially solid, made up of lines that are 0.6mm wide. The actual print will be made up of lines that are only about 0.44mm wide with a gap between them, because it's not possible to get 0.6mm width out of a 0.4mm nozzle.

**Expected results**
Cura's Preview should show the 0.44mm wide lines with empty gaps.

**Additional information**
A good photo of this can be seen in this reddit link:
https://www.reddit.com/r/3Dprinting/comments/7ndrur/printed_at_300_initial_layer_line_width_it/

Ghostkeeper, 2019-09-30:
When the initial layer line width is set to 150%, Cura will command the printer to extrude 1.5 times as much material and space the lines 1.5 times as far apart. That your printer is unable to handle this request is a limitation on your printer's side, not on Cura's side.

The printer is a physical thing that is limited by hardware, configuration and environment. In your case, it could be that the bed is too far away and the plastic needs to fill a bigger void than expected. Or it could be that the bed is too close and the plastic doesn't get room to expand sideways. Or it could be that the feeder is not able to provide enough force to push that much plastic out. Or that the temperature is too low and the material too viscous to spread out that wide. Or maybe the flat of the nozzle is not wide enough and it's expanding upwards as well rather than sideways.

You can set the line width to be 2 metres and Cura will display it as such, but that doesn't mean that the printer is able to achieve that. Normally we give a warning value when we expect the print to fail. In our experience, up to 150% of the nozzle size should be fine. Then again, I'm working with Ultimaker printers here that have fairly strong feeders because they are mounted on the side of the printer rather than on the print head. We've based our warning values on that. If your printer needs to give different warning values, the printer definition can override that. But I can't test that without your printer.

closed 2019-09-30 by Ghostkeeper

cmgreyhounds, 2019-10-01:
That makes sense. Is it ever possible to get 200% width or 300% width though?