Ultimaker/Cura #6445: Cura Profile reset

B14d3r11, 2019-09-29:
Every time I launch Cura instead of going to my Custom tuned profile for printing, it goes to the Default Standard Quality profile. How do I make it so it stays at my Custom Profile now?

Ghostkeeper, 2019-09-30:
I haven't been able to reproduce this bug. I'm testing this on Cura 4.3 since we broke something here on Master. These are my reproduce steps:

* Add Creality Ender 3.
* Set Infill Density to 42.
* Save the current changes as a profile named "bla".
* The new profile is then activated automatically.
* Restart Cura.
* The "bla" profile is still selected.

What are your reproduce steps? Perhaps you could fill in the [issue template](https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md)?

B14d3r11, 2019-10-01:
Okay, I did exactly as you said. It worked and the profile "bla" pops up correctly upon startup for the newly added Creality Ender 3 printer

If I select my current printer profile and select bla it also pops up on startup. But for some reason when I select my custom profile it doesn't want to stay? I've also noticed a slight visual change from the standard versus my profile.

Standard - https://i.imgur.com/cqWBHa9.png
Custom - https://i.imgur.com/8RE688E.png

Ghostkeeper, 2019-10-02:
So something seems to be different about that particular profile then. Do you know what is different? Was it created with a text editor maybe? Or was it created a long time ago? Maybe something has gone wrong in the version upgrade.

B14d3r11, 2019-10-03:
It was created with Cura 4.0, so I don't really know whats going on. I even had to recreate it in 4.3 cause it didn't transfer correctly...

CRCinAU, 2019-11-14:
I also notice that if you use the "Update" function to save the profile with the current settings, the profile will become corrupt and you'll get a new "empty" profile created.

Relaunching Cura will complain that the profiles are corrupt and prompt you to reset everything. No changes are made to the existing profile. As such, the print profile functions are pretty much useless in 4.3.

Ghostkeeper, 2019-12-03:
Can the profile in question be shared with us here so that we may be able to reproduce the problem then?

B14d3r11, 2019-12-06:
@Ghostkeeper I figured out it was the implementation of the Creawsome mod as I use the CR-10 as the basis for my Cura Profile for my X5SA. I uploaded the profile I used recently on a print and the supports are just thick as hell (picture linked for example)

https://media.discordapp.net/attachments/644695135007014928/646230961348018176/IMG_20191117_201215.jpg?width=1802&height=1352

[Coarse 75mms.zip](https://github.com/Ultimaker/Cura/files/3929778/Coarse.75mms.zip)

closed 2020-06-19 by no-response[bot]