Ultimaker/Cura #6446: In preview mode make layer number label visible when print settings is expanded

oishii-code, 2019-09-29:
**Is your feature request related to a problem? Please describe.**
I always have the print settings expanded so when I'm previewing the layers using the slider, the layer number label is always hidden behind it.

**Describe the solution you'd like**
Move the layer number label to be above or below the layer preview slider so that it is always visible, even when the print settings is expanded.

**Describe alternatives you've considered**
Moving the print settings to the left, however I think this would waste space that could be better used to view the model.

**Affected users and/or printers**
Any users who never collapses the print settings.

**Additional context**
Cura v4.3.0

[image]

[image]

oishii-code, 2019-09-30:
Closed. I noticed the printer settings windows could be moved.

oishii-code, 2019-09-30:
I added some more info and reopened it in case anyone has any feedback.

Ghostkeeper, 2019-09-30:
Yeah I've seen this before (but can't find the original bug report). We do consider this a bug. It's a bit hairy to fix because it involves space claims by plug-ins.