Ultimaker/Cura #6447: [4.3] Brim for support still not generated

Schleichmichl, 2019-09-29:
**Application version**
4.3.0

**Platform**
Windows 10, Intel i7 GPU, 24Gb RAM

**Printer**
Custom

**Reproduction steps**
1) Create Custom FFF Printer (Reprap, Dual Extruder, 200mm x 200mm Buildplate)
2) Add stl file
3) Choose PVA at second extruder
4) Select extruder 1 as Extruder for this model
4) Enable support
5) Set Extruder 2 as support extruder for everything
6) Enable Support brim
7) Set Brim as Adhesion method
8) Enable Prime Tower
9) Slice

**Actual results**
Brim for support is not generated for certain locations. Thus, support structures will be teared of mostly while printing the second layer with the other nozzle.

See also issue #6185 
There it was mentioned that this is also checked!

[STL.zip](https://github.com/Ultimaker/Cura/files/3667164/STL.zip)

[image]

**Expected results**
Brim is generated completely

**Additional information**

Schleichmichl, 2019-09-29:
There is an additional problem when using brim: The support nozzle is not primed at the first layer. That result in almost always missing support structures. I would appreciate when the brim is printed also with second nozzle.

Ghostkeeper, 2019-09-30:
Here is a project file that automates the reproduction steps: [CFFFP_RightTop.zip](https://github.com/Ultimaker/Cura/files/3671043/CFFFP_RightTop.zip)

Ghostkeeper, 2019-09-30:
The support brim is generated underneath the support, and doesn't expand outside of the support. It describes this in the description of the setting. For an example how support brim is meant to look and work, see my settings guide here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support_brim_enable.md

Schleichmichl, 2019-09-30:
Oh sorry, forget that step. Yes, flip it once as Step 2a)

Thanks for the clarification. 
Nevertheless, the support structure of these highlighted holes need to be supported. 
When I increase the support density, the other holes behave as expected. The bigger holes will not increase it's first layer area.
If the algorithm works as expected, then small islands of supports should be checked and strengthend.

There is strange behaviour that is not following your explanation:
When "Brim only on outside" is unchecked, we have again support printed in the air, this time in the small holes. See screenshots :-(
[image]
[image]
[image]

Ghostkeeper, 2019-10-01:
Yes, we've seen this as well. There is a routine that removes tiny inner circles of brim or skirt:

https://github.com/Ultimaker/CuraEngine/blob/d01f1487c63f2ab8b883645e8ca5a312f3245807/src/SkirtBrim.cpp#L97-L105

Those lines generally don't add towards bed adhesion but only really add blobs to your first layer so they are removed. Normally the brim gets subtracted from the support but in this case it didn't un-subtract the removed tiny brim perimeters.

Ghostkeeper, 2019-10-01:
Now it did add the extra prime line around the brim though?

Schleichmichl, 2019-10-01:
*Brim of second nozzle: 
Yes, it was added depending on "Brim only on outside" or "Bed adhesion extruder". In a certain constellation, it will NOT prime the support extruder.  This is really a bad bug that costs the user a lot of time :-(

*Support brim:
When I understand you right, there should be support underneath the magenta (support extruder) extrusion.
From user point of view, the bed adhesion for support structures printed with PVA is THE essential part getting dual nozzle prints working flawlessly.
I would really appreciate when you find a way that tiny supports are bound to build plate when "Enable support brim" is active. I made a lot experience where changing other options like 
-"skirt" vs. "brim"
-"Brim only on outside" 
-"Support horizontal expansion"
has so much influence on the support brim.

In my opinion, this shouldn't be the case. 
For instance: I can increase or decrease the  horizontal expansion of the support, but the support brim shouldn't change, at least not below a given minimum where bed adhesion is not garanteed.

Ghostkeeper, 2019-10-01:
> When I understand you right, there should be support underneath the magenta (support extruder) extrusion. Which extruder should have printed there? I think, the "bed adhesion extruder" should be the right choice.

Yes, in your last example where the support was floating in mid-air, the support should have gone all the way to the bottom. It's support, so it's expected to be printed with the First Layer Support Extruder.

> From user point of view, the bed adhesion for support structures printed with PVA is THE essential part getting dual nozzle prints working flawlessly.

Most people expect the first layer of the support to also print with the support extruder, so we've made that the default. You can change that though, with the First Layer Support Extruder setting.

> I would really appreciate when you find a way that tiny supports are bound to build plate when "Enable support brim" is active. I made a lot experience where changing other options like
-"skirt" vs. "brim"
-"Brim only on outside"
-"Support horizontal expansion"
has so much influence on the support brim.

In my opinion, this shouldn't be the case.
For instance: I can increase or decrease the horizontal expansion of the support, but the support brim shouldn't change, at least not below a given minimum where bed adhesion is not garanteed.

Skirt vs. brim and Brim Only on Outside have no influence on the support brim. Support horizontal expansion expands the support and thus also expands the room that the support brim can fill, but doesn't influence the technique of the support brim itself. If we didn't change the support brim when using support horizontal expansion, the brim would no longer be attached to the support's side and be much less effective. You might as well just reduce the Initial Layer Support Line Distance then, if you want it to be denser in the centre.

I'll repeat that the support brim does NOT go outside of the support. It never did and it can't, because there is no space there.

Ghostkeeper, 2019-10-01:
@Schleichmichl have you tried turning the setting "Brim Replaces Support" off?

Schleichmichl, 2019-10-03:
@Ghostkeeper Yes, that works as a workaround in this case.
[image]
But this is really hard to find out :-(

When I now switch back to skirt, I have the same problem of small support standing free on the build plate. 
[image]

Abusing brim to nail some support down isn't the final solution, right?

Ghostkeeper, 2019-10-03:
Combining "Brim Replaces Support" disabled with "Brim Only on Outside" disabled will cause it to print a brim inside those holes as you desired.

Cura doesn't really have a feature of "connecting small pieces together". It's been requested before though.

Schleichmichl, 2019-10-03:
Thx.

Requesting an option "Connect small islands on first layer" again. ;-)

BagelOrb, 2020-07-08:
Could you edit the title to reflect the final feature request as which this ticket is interpreted? I'm having a hard time understanding what feature is requested here.

Ghostkeeper, 2020-07-10:
It's indeed a bit confusing because there are several different things discussed here. I went with the original bug report. Everything else will be considered off-topic.

Schleichmichl, 2020-07-11:
I'm with Ghostkeeper. Focus only on the missing brim.
But if this issue switches from bug to a feature, what about a positive formulation or user story?

I as a CURA user 
want to use brim for an object that requires support
in order to strengthen build plate adhesion for the object and it's support structures.

Addition: The brim should also strengthen support structures, not weakening them.
It's not getting a better print if the print fails because of failing support structures.
This is very annoying to the user.

BagelOrb, 2020-07-11:
I think I remember some code which explicitly filters out brim lines if they are not long enough. Can't remember why it was there, though.

Ghostkeeper, 2020-07-14:
Yeah, it was linked in this topic a bit higher: https://github.com/Ultimaker/CuraEngine/blob/d01f1487c63f2ab8b883645e8ca5a312f3245807/src/SkirtBrim.cpp#L97-L105

The reason it's there is this feature request: https://github.com/daid/LegacyCura/issues/685

BagelOrb, 2020-07-14:
Ah yes. I think it should just be tweaked. Remove a brim line if the area is smaller than 2 x nozzle_size^2. The overextrusion only definitely happens at really small lines. Currently it's way too violent for the original purpose. Currently it's removing lines which cover an area of 10 x nozzle_size by 10 x nozzle_size !

This is a 5 minute fix. I suggest we reduce the value of 100 (10x10) to something like 4.

zackkit, 2020-07-29:
I have a very similar problem. Part of the support brim is not produced.
Till the 4th layer, some support is generated in the air.
Here are the 1st and 4th layer screen capture.
[image]
[image]

Ghostkeeper, 2020-07-30:
Your patches are not exactly tiny though. Do you have a project file of that that you can share?

zackkit, 2020-07-31:
Sure!
Here is the stl: https://drive.google.com/file/d/1jZNbqKJwJw9roGfxOatiuDr2rTojZ3Q5/view?usp=sharing

Ghostkeeper, 2020-08-03:
Oh, that's not a project file. It's an STL model. What I meant is to go to File -> Save... and share the resulting file. It's a 3MF file with your entire scene and added information about which profiles and settings you're using. It's meant to be able to reproduce a slice exactly.

I can load your STL in Cura, but it doesn't reproduce the problem, probably because I'm using slightly different settings than you:

[image]

I also find it strange that support is yellow for you, by the way.

zackkit, 2020-08-16:
[image]
This is the 1st layer.
[image]
Then 2nd layer.
[image]
Till the 4th layer, the whole support finally shows up!
Here is the .3mf and .stl files.
I really have no idea what's going on.
Thanks for your help!!!
[3mf_and_stl.zip](https://github.com/Ultimaker/Cura/files/5080116/3mf_and_stl.zip)

smartavionics, 2020-08-16:
Using my Cura which is vaguely like 4.7, I see support there as expected.

[image]

Have you tried using a recent Cura release?

Ghostkeeper, 2020-08-17:
Yes, in the actual 4.7 state it's currently already fixed. Could be the fixes we made for the support stair stepping algorithm.

zackkit, 2020-08-18:
WOW!
I'm on 4.6.2 which is why I have this problem!!!
Thank you so much!!!
After I switched to 4.7.0 the problem is gone.

closed 2024-12-07 by github-actions[bot]