Ultimaker/Cura #6448: Incorrect spports and supports in the air

FlyToha, 2019-09-29:
**Application version**
Cura from 3.6.0 to 4.3.0

**Platform**
Windows 7 x64, Core i 7 4930k, 16gb, dual gtx760

**Printer**
Ender 3 Pro
**Reproduction steps**
Generation supports

**Actual results**
I worked with Cura about 4 monts and all was fine. I anytime used Cura profile by FDG (Official_FDG_Cura_Ender3), because with this profile i had rly cool result on every printed part. But yesterday i did try to prepare another model and when i pressed "generate supports", i saw, that supports placed around model, but they do not touch the model. Next i tried other model, which was printed fine before and i saw, that supports also don't touch part and even "flying" inside holes of model. Its happened on Cura 4.2.1. I tried to full reinstall Cura, tried 3.6.0, 4.1.0, 4.2.1, 4.3.0 - problem still exstis. I read some issue topics here, and tried to set minimum support area to 0, minimum x-y distance to 0.1, and x-y distance to 0.4 and other values. Of course, i also tried to switch profiles, and i tried to use standart Cura profiles - nothing helped. And final: i thought, that maybe i have problems with my pc (some drivers conflict, os issue and etc) and i did full reinstall Windows - did not helped.
**Expected results**
Supports should follow the mesh and they should touch the mesh at selected overhang angle.

**Additional information**
[image]
[image]
[image]
[image]
[Models_and_Prof_SupportsIssue.zip](https://github.com/Ultimaker/Cura/files/3667703/Models_and_Prof_SupportsIssue.zip)

Ghostkeeper, 2019-09-30:
Reduce your Support Minimum Area: duplicate of https://github.com/Ultimaker/Cura/issues/6345

FlyToha, 2019-09-30:
> Reduce your Support Minimum Area

I tried it and it's not fixed flying supports
[image]

Ghostkeeper, 2019-09-30:
Right, no it doesn't influence support resting on the model but on a very steep slope. This support rests on the model but due to your settings for the X/Y distance (0.7mm) it keeps a large distance so it'll effectively just fall down.

My preferred method for this type of support is either tree support (PLA) or conical support with negative angles (PVA). Or to filter it out with support blockers.

closed 2019-10-01 by Ghostkeeper