Ultimaker/Cura #6452: 4.2/4.3 keyboard crippled

strikingcrayon, 2019-09-30:
**Application version**

4.2.? and 4.3. I was only using 4.2 for a day or two before 4.3 came out.

**Platform**

Wins 10 Pro.

**Printer**

 Anycubic i3 Mega

**Reproduction steps**

I have no idea how you will reproduce this. 
Approximately 30-40% of the time that I try to type while Cura is open my keyboard ceases to function normally. Keys will "stick" and type huge strings of letters or keystrokes will not be registered at all. 
This began when I updated to whichever Cura was right before 4.3. 4.2.?, I don't recall the last digits. Up until then, I was using an older version that came packaged on an SD card with my printer without issue, Though I don't remember which edition that was. I've reinstalled windows twice (clean installs) and even reflashed the firmware on my keyboard (Kono Knightfox). It isn't a system performance issue as it happens both when slicing, and when not, with no seeming preference. I am beginning to suspect that it happens most often when the Cura application is focused though not exclusively. The issues do persist when I focus others windows. However, I think once the issue stops when I'm in other software, it doesn't return until I refocus Cura.

**Actual results**

 There weren't any steps to give so I just gave an explanation above. The results are always the same,

**Expected results**

Keyboard behaves wacky. My system will receive massive amounts of never-ending keystrokes, often followed by no keystrokes being received at all. It's always the sticky keys first which makes the lack of keystrokes seem like delayed ghosting. The ghosting will be random for a while without most sticky keys, but eventually, the sticky keys come back, or the problem goes away for a while. 

**Additional information**

I guess computer things and my current diagnostic attempts? I have a 4790K CPU with 32GB of ram, windows 10 pro running off an NVME SSD, and if it matters a 1070 GPU. The keyboard I'm using when the problems happen is a Kono Knightfox which I assure you has undergone the most rigorous of my early testing to rule out. The issue doesn't happen on other computers with the same versions of Cura. I've reinstalled Cura multiple times. Even reflashed the firmware on my keyboard. The only way it has anything to do with the keyboard at this moment is if the keyboard is only part of a multi-step cause. 
From a layman's side, I'm at a loss for other diagnostic measures. It's a peculiar issue.

Thank's for your time.

Ghostkeeper, 2019-09-30:
Perhaps your keyboard is registered via a COM port and Cura interferes with that?

Could you try these steps:
* Open Cura.
* Click on "Marketplace" in the top right.
* Click on "Installed" in the top right of the pop-up.
* Find the plug-in called USB Printing, about a third of the way down the list. Uncheck the checkmark in front of it.
* Restart Cura. See if the problem still exists then.

strikingcrayon, 2019-09-30:
I have done as instructed however I won't be able to test it until later this evening. I'll reply tomorrow with my results. Thanks.

strikingcrayon, 2019-10-02:
This has fixed the problem @Ghostkeeper ! Thanks! 😃 

Sorry for the delay but I didn't end up using Cura enough on the first day to be sure. 

I'm guessing this fix means I can't use Cura and print over USB with my current setup then?

nallath, 2019-10-03:
Your keyboard is getting confused from Cura attempting to talk to it as if it was a printer. 

What you could try to do is change some settings in windows so that it's no longer being recognised as a COM port (and only as a keyboard)

closed 2020-06-19 by no-response[bot]