Ultimaker/Cura #6453: 4.3.0 Fails to Import Profile

UberEgo, 2019-10-01:
Looks like it may be a regression based on #6033 

Environent
4.3.0
OSX 10.14.4
CR-10 

Cura appears to fail to import all or part of configuration files with the following error:
"Could not find a quality type fast for the current configuration."

[profiles.zip](https://github.com/Ultimaker/Cura/files/3674112/profiles.zip)

The first profile in the archive fails to import at all, while the second (priest) profile is created, but it appears that not all of the settings may be imported correctly.

Ghostkeeper, 2019-10-01:
Yes, for the `Cura_CHEP_CAT.curaprofile` file this is expected behaviour right now. The quality level `fast` that the profile was originally based on doesn't exist any more. That's why it gives an error. It can't reconstruct your profile.

The `Priest3.4.1V8.curaprofile` file gives the same error and I'd expect it to not get added to Cura then. So that's a bug.

UberEgo, 2019-10-01:
If I were to awk the profile and replace fast with an existing profile, would it work?
If so, could we prompt to pick a new base profile?

UberEgo, 2019-10-01:
I'm not sure of the reasoning behind the serial encoding of the profiles. 
It really makes them non-portable and only editable within the application.
It may be a worthy improvement to move to a YAML format -- unless there is a discussion on this that I have already missed.

Ghostkeeper, 2019-10-02:
> If I were to update the profile and replace fast with an existing profile, would it work?

Yeah, I expect so.

The .curaprofile files are zip archives with INI files. You can open it with an archiver application and then edit them with a text editor. Originally they were the pure INI files themselves but then we needed to support multi extrusion. Packing them in a zip archive was a very effective way to deal with that.

> If so, could we prompt to pick a new base profile?
Or provide a legacy profile mapping option --

We have a legacy profile mapping. This happens on any import from an older version behind the scenes, and we use it also to rename renamed settings and such. Only what happened for your printer is that in version 4.1 to 4.2 the printer profiles were changed such that Cura considers it a completely different printer and the matching is not applied. For this reason we started allowing importing profiles from different printers in 4.3. But only for the configurations that were still available in the new printer. It's a bit overzealous to make a dialogue to change the material, nozzle, build plate and quality level if they aren't available any more in the printer you're importing it for. I can't justify spending time on implementing that, sorry!

nallath, 2019-10-02:
The bug is fixed in f5f91c9c3a3086b5b576fc2e2907454c8e1d892d

UberEgo, 2019-10-02:
Thanks very much for the feedback.
Appreciate it.

Ghostkeeper, 2019-10-03:
So now it no longer partially adds profiles if it is discovered later that some profiles are incompatible. Everything behaves like the `Cura_CHEP_CAT.curaprofile` file. Thanks for that bug!

closed 2019-10-03 by Ghostkeeper

photodude, 2019-10-08:
I also came across this with the [CHEP 4.2.1 profiles](https://www.chepclub.com/cura-profiles.html)
opened `.curaprofile` Zip -> edit the two ini files
changed `quality_type = super` to `quality_type = normal`

and the profiles would load.

ddbrown30, 2019-11-06:
Is there an ETA on when this fix will be released?

Ghostkeeper, 2019-11-07:
We've just uploaded the beta for Cura 4.4. The fix should be in that release. The stable is due in 2 weeks (currently).

Greaga, 2019-11-07:
4.4 still doesn't allow the ability to import the profile for the Anet ET4. still says same error.

Ghostkeeper, 2019-11-07:
Nice, works fine then? It's expected to give an error when adding a profile that is based on a quality profile that doesn't exist. Because we can't load that quality profile then, since it doesn't exist.

The fix here is that if that error occurs, it's not secretly still adding a broken profile, as with the `Priest3.4.1V8.curaprofile` file originally.

ddbrown30, 2019-11-07:
Ah, seems I also misunderstood the fix. I thought that this was fixing backwards compatibility.

Ghostkeeper, 2019-11-08:
The point is that in version 4.3 Cura started to allow you to import a profile that was made for a different printer. This was not previously possible. We decided on this because the change in Creality printers in version 4.2 made Cura consider those printers a "different printer".

This error message occurs when you import a profile into a printer that doesn't have the quality level that the original profile was made for. It has nothing to do with backwards compatibility.

ddbrown30, 2019-11-08:
Maybe I'm misunderstanding you, but from what you're saying it sounds like a profile for a printer in 4.2 should still work for that same printer in 4.3. I'm attempting to use the profile below with my Ender 3 and it's not working. Granted, it has several printers listed here, so maybe that's the issue.

https://www.3dprintedtabletop.com/download/cura-4-2-and-higher-3dprintingpro-mini-terrain-profile-creality-ender-3-ender-5-cr-10-ender-2/

magiccak, 2019-11-09:
I'm trying to import the very same profile too, on 4.3.0, without a success and getting the same error. Is there any solution yet?

> Maybe I'm misunderstanding you, but from what you're saying it sounds like a profile for a printer in 4.2 should still work for that same printer in 4.3. I'm attempting to use the profile below with [...]

ddbrown30, 2019-11-10:
> I'm trying to import the very same profile too, on 4.3.0, without a success and getting the same error. Is there any solution yet?

I managed to figure out a solution. As it mentions on the site, those settings are derived from the profile at Fat Dragon Games. If you head there are follow the guide, it will actually have you install a legacy printer into Cura. This printer supports both the profile above, as well as the Siepie profile, if you want to use it.

http://www.fatdragongames.com/fdgfiles/?p=4934

sebachu, 2020-01-01:
Hi everyone, I was able to fix this problem by manually 'exporting' all my custom profiles from Cura 4.3, and then manually importing all the profiles into Cura 4.4.1. Not sure if it will work for everyone, but it worked for me (on an Ender 3). Cheers.

hypercube33, 2020-05-19:
> Nice, works fine then? It's expected to give an error when adding a profile that is based on a quality profile that doesn't exist. Because we can't load that quality profile then, since it doesn't [...]

Software gore. Why doesnt it just say hey, this one thing isnt right here are the available options and then EDIT the file as its imported with the users selection - warning that it may not be accurate or work. This is pretty lame.

poikilos, 2021-03-04:
It is still happening on 4.8 :(

BarsMonster, 2021-03-28:
I also faced this issue in 4.8.0. Unfortunately, changing quality_type to standard or normal or Normal does not fix it. 
It is very non-intuitive.

SuddenlyTurtles, 2021-06-06:
Bars commented that changing is did not work but I just got it working by setting quality_type = standard. Try this out.

[3DPrintingProMiniFile-Cura-4.2-and-on.curaprofile.zip](https://github.com/Ultimaker/Cura/files/6603541/3DPrintingProMiniFile-Cura-4.2-and-on.curaprofile.zip)

SuddenlyTurtles, 2021-06-06:
> 
> 
> I also faced this issue in 4.8.0. Unfortunately, changing quality_type to standard or normal or Normal does not fix it.
> It is very non-intuitive.

I just got it working by setting quality_type = standard. Try this out file out.
[3DPrintingProMiniFile-Cura-4.2-and-on.curaprofile.zip](https://github.com/Ultimaker/Cura/files/6603542/3DPrintingProMiniFile-Cura-4.2-and-on.curaprofile.zip)

camelator, 2022-01-25:
same issue here

ChumKiu, 2022-07-22:
You can try by changing also the setting_version into the metadata section.
The value shall match the one generated by the used Cura, if you try to import a file that have a different setting_version you got the error above.
Hope it help!

sclarady, 2023-05-06:
For me I had to change the Quality to standard and the setting_version.

belobig, 2023-05-11:
On Cura 5.3.1, I was attempting to import Cura profiles from 5.0 for an Anycubic Kobra Max. I discovered that for quality_type, neither 'normal' nor 'standard' would work. I dug into the config files for the printer (for me, they were in AppData\Roaming\cura\5.3\quality_changes) on the computer with the newer version of Cura, and discovered a quality_type setting there of 'pla'. So I changed the quality_type setting in the 2 text files for the profile I was trying to import, then recompressed them, and changed the filetype back to .curaprofile (as described above by @Poikilos ), and was able to successfully import the profile. Thank you all for your input that got me this far!

khashashin, 2024-04-28:
> It is still happening on 4.8.0 :(. The problem is that only certain printer profiles such as Creality CR-10 have the old "standard" setting. I found a complex workaround:
> 
> * Rename .curaprofile to .zip
> * Extract the two files
> * Edit both files (each profile contains two text files) by hand to change "standard" to "normal"
> * Recompress them and rename the .zip back to .curaprofile (You'll need to turn on "always show extensions" or use Command Prompt to change known extensions if using Windows, adding .curaprofile will not work otherwise and it will become .curaprofile.zip which will not work)
> 
> This is not an acceptable workaround clearly, since I'm the only person who managed to do that this many versions later. We can't do this for every profile like that. Cura needs to be fixed, such as by letting you choose a quality type from a list of valid ones. I've submitted the changes to fatdragongames so they or I will add a comment here if they upload the 3 changed profiles.

Thank you! I did it as in your instruction but changed the quality_type to standard (Cura Slicer Version 5.7.0)