Ultimaker/Cura #6455: [4.3] and older: Print sequence of skirt and prime tower brim

Schleichmichl, 2019-10-01:
**Application version**

4.3
**Platform**

Windows 10

**Printer**

[PrimeTowerbrimBeforeSkirt.zip](https://github.com/Ultimaker/Cura/files/3675296/PrimeTowerbrimBeforeSkirt.zip)

**Reproduction steps**

1) Open attached project file
2) Slice

**Actual results**

Brim of prime tower is printed first

**Expected results**

Skirt must be printed first

The Skirt is there to prime the nozzle. Also, you can check if print bed calibration is fine for this job. The brim of the prime tower should stabilize the tower. If it is only printed partly, this could get a problem in the end of the print, thus wasting a lot of time, material and energy.

**Additional information**

[image]

nallath, 2019-10-04:
We're looking into it to see if we can find a quick fix.

closed 2019-10-15 by Ghostkeeper

Ghostkeeper, 2019-10-15:
Thanks for the report!