Ultimaker/Cura #6457: ModifySettingsforOverlap does not work for Horizontal Expansion setting

daT4v1s, 2019-10-01:
**Application version**

4.3.0

**Platform**

Linux Ubuntu LTS 18.04 Intel Graphics

**Printer**

ultimaker 2+

**Reproduction steps**

1. import stl with thin text
2. apply a horizontal expansion value of -0.1mm
3. since the text is now mangled/corrupted, exclude horizontal expansion where the text is
by adding an object, that envelops just the text-part of the shape
4. Set the new object's MeshType as ModifySettingsforOverlap, and add custom settings HorizontalExpansion, and set to 0.0mm

**Actual results**

the text is still corrupted, the local settings are not followed, only the global ones

**Expected results**

HorizontalExpansion of -0.1mm,
 except where the ModifySetting block that envelops the text is

**Additional information**

[image]

Ghostkeeper, 2019-10-02:
Yes, right now we apply horizontal expansion before the special mesh types. This way the settings can be changed of part of the expanded mesh. Like this:

[image]

We can support one or the other, not both.

Ghostkeeper, 2019-10-15:
While we think that the suggested precedence would be more useful than the current one, we don't consider it very important to change it around now. Let's defer this to a later time.

daT4v1s, 2019-10-20:
workaround: 2 meshes with different settings
turn off AutomaticallyDropMesh (preference setting) to allow for positioning

pyrokidd89, 2020-11-10:
I have this issue also. I would like to apply horizontal expansion to the first ~4 layers to reduce elephants foot. Can you explain more about this 2 mesh workaround?

Ghostkeeper, 2020-11-17:
What daT4v1s meant is to actually break up the 3D model (e.g. STL) into 2 model files. One model file is the large structure of the model, the other is the text. Then apply the horizontal expansion only on the big model. In your case, you could cut your model into 2 parts: One thin model that is just the bottom 4 layers, and another model for the rest of the print. Then stack them on top of each other (perhaps with Ctrl+G as an actual merge like one would for dual extrusion) Then you can adjust the horizontal expansion differently for those two models.

We have a different workaround for the original use case here now as well, with a separate Horizontal Expansion setting for holes.

closed 2023-06-13 by github-actions[bot]