Ultimaker/Cura #6460: Add Printer Creality Ender 3 PRO

linuxjunkie999, 2019-10-01:
**Is your feature request related to a problem? Please describe.**
Adding a new printer that is explicitly the pro version is one way to solve this bug https://github.com/Ultimaker/Cura/issues/6459

**Describe the solution you'd like**
The Ender 3 PRO is an improved version of the Ender 3 and therefore has some specs better and or different than the original.

**Describe alternatives you've considered**
I tried changing the ender 3 printer definition to the specs of the ender 3 pro but did not succeed.

**Affected users and/or printers**
Anyone who has an Ender 3 PRO and wants to use it to its full potential.
 
**Additional context**
_this is actually a .ini file_
[Ender-3 Pro.txt](https://github.com/Ultimaker/Cura/files/3677747/Ender-3.Pro.txt)

[Ender-3 Pro Reference data.pdf](https://github.com/Ultimaker/Cura/files/3677748/Ender-3.Pro.Reference.data.pdf)

rburema, 2019-10-02:
We're wholly dependant on the community and/or other manufacturers when it comes to profiles. In other words: pull requests welcome!

tedukes, 2019-10-02:
I have an Ender 3 Pro. Its shows up as an Ender 3. You can go into Manage Printer >Printer > Machine Settings sand adjust the dimensions. Not sure what else you would need to change with regards to Cura. Would be interesting in knowing because that's all I've done.

rburema, 2019-10-02:
@tedukes What _could_ be done (but in cases like these where there are many similarities doesn't necessarily have to, provided you're happy to change the machine settings) is that a separate profile is made for the 'Pro' (perhaps inheriting from the 3 or a mutual base class if you want to be proper about it), so people won't _have_ to go into the machine settings, but can select 'Ender 3 Pro' from the settings menu.

linuxjunkie999, 2019-10-07:
> @tedukes What _could_ be done (but in cases like these where there are many similarities doesn't necessarily have to, provided you're happy to change the machine settings) is that a separate profile [...]

[Ender-3 Pro.txt](https://github.com/Ultimaker/Cura/files/3698014/Ender-3.Pro.txt)
[Ender-3 Pro Reference data.pdf](https://github.com/Ultimaker/Cura/files/3698016/Ender-3.Pro.Reference.data.pdf)

linuxjunkie999, 2019-10-07:
Directly from Creality the non pro info 

[Ender 3.txt](https://github.com/Ultimaker/Cura/files/3698053/Ender.3.txt) [actually an ini file]
[Ender 3 Reference data.pdf](https://github.com/Ultimaker/Cura/files/3698054/Ender.3.Reference.data.pdf)

linuxjunkie999, 2019-10-07:
From what I can tell the ender 3 pro is a significant revision to fix bad design choices in the original version.

linuxjunkie999, 2019-10-07:
You can get the information by downloading the SD card contents  from one of creality's websites (they keep changing where to get the actual files...)

linuxjunkie999, 2019-10-07:
[image]

Ghostkeeper, 2019-10-08:
Those are nice files, but they are not the printer definitions we'd need. They are made for a version of Cura that was released before Cura even knew that different materials existed. We can import them, but we can't ever test their workings without buying an Ender 3 Pro.

tedukes, 2019-10-09:
Sure am glad i didn't spend $6K for an Ultimaker and have to rely on Cura. Get better prints with Prusa Slicer. Cura does have features regarding supports and use only for that. My E3 Pro with about $300 in upgrades, SKR 1.3, 2208's UART, dual Z axis, direct drive extruder, Micro Swiss hotend come in way less than an Ultimaker.
Need to fix the bugs before adding new features!! Like tuning a printer, do one thing at a time! That's why I run RHEL 7 servers and not some bleeding edge OS full of bugs and vulnerabilities!! 

Glad i was able to find Cura 3.6 as they have removed all previous version from their website.

nallath, 2019-10-09:
@tedukes What does that have to do with anything mentioned in this ticket?

linuxjunkie999, 2019-10-11:
> Those are nice files, but they are not the printer definitions we'd need. They are made for a version of Cura that was released before Cura even knew that different materials existed. We can import them, but we can't ever test their workings without buying an Ender 3 Pro.

Well you could compare the ender 3 ini and ender 3 pro ini for differences to see if any of the differences are relevant to making a proper printer definition...

linuxjunkie999, 2019-10-11:
My hope was that you would compare the ender 3 printer definition cura has and the two profiles I provided to make an ender 3 pro printer definition...

Ghostkeeper, 2019-10-14:
We can compare them from theory, but printing in practice is always different.

kazooless, 2019-10-22:
@linuxjunkie999  It is my understanding the Cura developers in charge of this actually work for Ultimaker. They don't have all the other brand printers out there to test and write code for. They add what is given to them by the community if the community is willing to put their time and effort in. Being that they give their software away for free, I am very grateful they do that and I understand why they don't do the specific work on configurations for competitors machines.

That said, if you want to compare the settings in the profile you share above for the pro with the current built-in Ender-3 profile (found here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_ender3.def.json and here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_base.def.json ) I would be happy to add an Ender-3 Pro profile to my fork of Cura and then submit a pull request for you. Just give me the settings that are either different or don't exist in the two files I link above and I'll get that done for you.

Ghostkeeper, 2019-10-22:
If it helps, we have a translation mapping from profiles for Legacy Cura 15.04 to profiles for Cura 2.1, here: https://github.com/Ultimaker/Cura/blob/master/plugins/LegacyProfileReader/DictionaryOfDoom.json

CRCinAU, 2019-11-14:
Ender 3 Pro owner here - there's sweet FA difference between the 3 and 3 Pro when it comes to slicing an operations.

You can close this as the Ender 3 definition works fine for the Ender 3 Pro... Slicing doesn't care about a different power supply, 2040 extrusions instead of 2020....

closed 2019-11-26 by linuxjunkie999