Ultimaker/Cura #6462: 4.3.0 Bring back Maximum Z Speed.

DarkAlchy, 2019-10-01:
Seems you removed this VERY valuable setting (this has nothing to do with Z hop speed) and now that it is gone I know a lot of us with fine pitched leadscrews are not able to upgrade due to that missing.

Ghostkeeper, 2019-10-01:
As discussed [here](https://www.reddit.com/r/Cura/comments/dbq6fw/missing_maximum_zspeed_in_43/f24hiww/), the Maximum Z Speed setting was replaced by Z Hop Speed in version 4.2, but still affects the speed at which layer switches are done as well. So yes, it has something to do with Z hop speed. And nothing really changed with this except that the Z hop speed is now a fixed number by default rather than the maximum speed that the firmware can do.

I do think that the speed for layer switches is an unrealistic setting to achieve. Even with a big layer height like 0.3mm, you'll have 0.15mm to accelerate to your supposed Z Hop Speed of 60mm/s. This needs an acceleration of [12000mm/s^2](https://www.wolframalpha.com/input/?i=0.15+%3D+0.5+*+x+*+%2860+%2F+x%29%5E2). For comparison, the maximum acceleration of the Ultimaker 2 in the Z direction is 100mm/s^2. So most likely the top speed in there is going to be on the order of 0.9mm/s or so with Jerk, rather than 60mm/s.

DarkAlchy, 2019-10-02:
No, I just did a test and I set my Z-Hop (I had to turn it on first) to 4mm/s and STILL I see this on my layer changes - G0 F3600 X84.79 Y132.845 Z0.3

See that?  Z-Hop off or on that Z is riding along with the X and Y speed which is insanely over bloated.  If you think Z-Hop is affecting layer change speed as well this proves it is not.  The actual Z-Hops do have the F240 so my 4mm/s.

Ghostkeeper, 2019-10-02:
That's strange. I'm unable to reproduce that though. Here's a snippet of what I get when I set the Z hop speed to 4mm/s, at every layer change (lowercase comments added by me):

```
G1 X111.675 Y112.325 E74.16806 ; <-- Last extrusion of previous layer.
G0 F15000 X111.675 Y112.125
G0 X112.06 Y111.94
G0 X112.3 Y111.7
;MESH:NONMESH
G0 F240 X112.3 Y111.7 Z5.67 ; <-- Z move explicitly done at 4mm/s.
G0 F15000 X120.34 Y111.34
;TIME_ELAPSED:415.535789
;LAYER:54
;TYPE:FILL
;MESH:cube.stl
G1 F4178.6 X120.339 Y106.47 E74.20013 ; <-- First extrusion on new layer.
```

Could you maybe then fill in the [bug template](https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md) and provide a project file that reproduces your bug?

smartavionics, 2019-10-02:
> No, I just did a test and I set my Z-Hop (I had to turn it on first) to 4mm/s and STILL I see this on my layer changes - G0 F3600 X84.79 Y132.845 Z0.3

Yes, but that speed (60 mm/S) is for the vector that is from the current X/Y/Z position to the position specified by the G0, so the Z speed required could be very low if the X & Y distances are appreciable.

DarkAlchy, 2019-10-02:
@smartavionics No matter what it is Z cannot be allowed to have 60mm/s on it.

DarkAlchy, 2019-10-02:
@Ghostkeeper A template?  All I have is the gcode file with and without a z-hop.

Ghostkeeper, 2019-10-03:
Sorry, let me ask you this then: How did you create that g-code file? Using Cura, I suppose, but what steps did you take? Change any settings, load certain models, etc?

And secondly, if you were using Cura to create this g-code, could you just before clicking "Save to File", go to File -> Save, put the resulting .3mf file in a zip archive and upload it here on Github? That should allow me to reproduce the issue.

DarkAlchy, 2019-10-03:
[CFFFP_Boardholder2a.zip](https://github.com/Ultimaker/Cura/files/3686433/CFFFP_Boardholder2a.zip)

Ghostkeeper, 2019-10-03:
That's not a project file :disappointed: Just a 3MF file. Don't export, just save!

DarkAlchy, 2019-10-03:
Wait, I never exported I hit the button to save and moved the drop down to 3mg (was on stl) and did it like that.

Ghostkeeper, 2019-10-07:
It's confusing, sorry. What I mean is to go into File -> Save... , also not to press the blue button that you'd normally use to save the g-code and such.

kazooless, 2019-10-07:
This seems to be a problem for me too. I've got a new DIY printer and I've got way too much gear translation on my Z-axis right now (In the process of rectifying but for now...) 

In v4.0, if I set maximum z speed to 2mm/s my print works fine. If I set it to 3 or higher, the axis won't move at all. 

In v4.3, using the "printer settings" plugin, I can change max speed for z and it does nothing. The print fails no matter what I enter there.

smartavionics, 2019-10-08:
Have you tried adding "M203 Z2" (or Z120 if your firmware expects the max speed to be in mm/min) to your start gcode? That would limit the z speed to 2 mm/s just like Cura did before this recent change.

kazooless, 2019-10-08:
@smartavionics I have not done that. I only just figured out the problem today and went looking to see if there was a bug already reported. I simply sliced in 4.0. I can give that a try though.

closed 2020-06-19 by no-response[bot]