Ultimaker/Cura #6464: Cut and Paste start Gcode file

Mark-Blanton, 2019-10-02:
Cannot cut and paste a custom Gcode start file into Cura 4.3 when running Cura as an appimage in Linux. The last time I could do this was in Cura 4.0.0.

Ghostkeeper, 2019-10-02:
I'm unable to reproduce this. I can cut, copy and paste in the g-code fine using the hotkeys Ctrl+X, Ctrl+C and Ctrl+V.

Mark-Blanton, 2019-10-02:
Up through 4.0.0 I could open my custom Gcode file (created in LibreOffice Writer) and simply highlight the contents of the file with my mouse, click on "Copy",  open the machine settings in Cura, highlight the current gcode start file contents then click and paste the new contents into the file, overwriting the old contents, using my mouse.  I can't do than any more.  I cannot paste the new start code file into Cura with my mouse.  I have tried this in Ubuntu 19.04, Debian 10, Ubuntu Mate 19.04, Linux Mint, and in ArcoLinux which I am currently running.

I tried using the hotkeys as you suggested and that worked!  Thank you I consider the issue closed.

closed 2019-10-02 by Mark-Blanton

Ghostkeeper, 2019-10-03:
Ah, right. It's a duplicate of https://github.com/Ultimaker/Cura/issues/6159 then. Ctrl+C and Ctrl+V still work.