Ultimaker/Cura #6465: [FR] Make "Infill Layer Thickness" behave like "Support Z Distance"

kulfuerst, 2019-10-02:
**Is your feature request related to a problem? Please describe.**
The Infill Layer Thickness has to be updated with every change I make to the Layer Height. If I load one of my profiles and change the Layer Height I always have to pay attention to the Infill Layer Thickness as well because it doesn't automatically round up like Support Z Distance does, which can be set once and then never be bothered with again.

**Describe the solution you'd like**
Make Infill Layer Thickness automatically round up to the next multiple of the Layer Height, the same way the Support Z Distance does. That way it can be set once and then never be bothered with again. Small layers would automatically print with thick infill saving lots of print time while large layers print with infill of identical thickness.

**Describe alternatives you've considered**
Always change the setting by hand, which has cost me hours of print time because it's very easy to forget.
Alternatively, the implementation of a modifiable check box of settings that should be checked before an object is sliced, this would however be very tedious for the user. 

**Affected users and/or printers**
Anyone who ever changes their Layer Height

**Additional context**
[image]

Thank you for your support!

Ghostkeeper, 2019-10-02:
I've made one change immediately: https://github.com/Ultimaker/Cura/commit/672fc58930052749c497862242f7871c888ac223

So for your example it would work (coincidentally).

Both Infill Layer Thickness and Support Z Distance round to the layer height. However the difference is that for Support Z Distance it is allowed to have a distance of 0, so changing the layer height never causes it to go into an invalid state. For Infill Layer Thickness, a thickness of 0 layers is not allowed. Previously it would disallow anything lower than the layer height but I've changed that now to disallow anything lower than half the layer height, because above that it gets rounded to one or more layers anyway and it would still slice.

Making it round up instead of round to the nearest integer is a change that would change every existing profile currently in circulation. I could bring that up, but I don't think it's worth the hassle for the profile designers.

kulfuerst, 2019-10-02:
@Ghostkeeper 
Okay I've changed that line and it appears to work. The behavior is extremely difficult to predict though. I'm trying to wrap my head around how it is deciding the layer height of the infill now but I can't really make much sense of it. Let me explain:
I wanted to always print infill at twice the layer height when printing at layer heights of 0.08, 0.12 or 0.16, and print infill identical to layer height at layers 0.2, 0.24, 0.28 & 0.32. Now with this change I can produce the desired behavior when setting the infill layer height to 0.18! I expected this to work at 0.161 for all it's worth. Now I'm just trying to understand why 0.18 is now the magic number and if this has any unwanted consequences (e.g. not adjusting the flow to the actual infill layer thickness but always printing with a flowrate adequate for 0.18 mm layers).
Do you already understand this behavior? Can you explain to me why 0.18 produces the desired behavior now? Thank you very much for your help!

kulfuerst, 2019-10-02:
It appears to be working fine. I'm just glad I don't have to write the text for the mouse-over that explains the behavior of this function :D
My test prints at 0.16 mm layer height took less time than printing the same object at 0.2 mm layer height. Printing at 0.12 took just barely longer than 0.2. That this is now automatic will save me a lot of time!!! Thank you very much for this :)

closed 2019-10-02 by kulfuerst

Ghostkeeper, 2019-10-03:
I'd expect that if you set the infill layer height to 0.24mm and the normal layer height to 0.08mm, you'd get infill layers at 3x the layer height. That's also what happens when I try that. Are you sure you counted correctly?

With 0.12mm normal layer height I'd expect that your infill layers will be 2x the layer height.

With 0.16mm the 0.24mm infill layer height would be exactly in the middle between 1 layer (0.16mm) and 2 layers (0.32mm). I'd expect it to round up then to 2 layers, so your infill layers would become 0.32mm thick.