Ultimaker/Cura #6466: The timing label partialy hides the layers bar in monitor mode

AMI3, 2019-10-02:
**Application version**

**Platform**

cura 4.3.0
windows 10

**Printer**

**Reproduction steps**

move from prepare to plan tab

**Actual results**

the program calculated the time it would take and showed it on an unmovable label, which partialy hides the layer bar

**Expected results**

**Additional information**

screenshot added
[image]

Ghostkeeper, 2019-10-03:
Yup, I see it. It seems to happen if the window is smaller about the height of an 11" screen or so.

Liger0, 2019-10-08:
Also related to the GUI, the layer number is hidden by the profile settings.

closed 2024-12-07 by github-actions[bot]