Ultimaker/Cura #6468: [4.3.0] Hang when saving g-code or printing over network with stretch script enabled

ryanshow, 2019-10-02:
**Application version**
4.3.0

**Platform**
Windows 10 / Linux (Ubuntu 18.04.3 LTS)

**Printer**

Ultimaker S5

**Reproduction steps**

- Load model
- Enable "post stretch script" in Extensions->Post Processing->Modify G-Code
- Slice
- Attempt to save G-code or Print over the Network

**Actual results**

Cura hangs and will not recover. The g-code is never saved and the print is never started.

**Expected results**

The g-code should save or the print should start. Cura should not hang.

**Additional information**

I don't know exactly how this bug is triggered, as I was able to successfully print over the network with the post stretch script enabled several times when I was first experimenting with this feature. Then for some reason, cura started to hang whenever I would attempt to print or save gcode, regardless of what model I loaded or what settings I selected or what operating system I used. I was only able to print again once I disabled the post stretch script.

nallath, 2019-10-03:
Do you happen to have the log files of when this happens? I think the issue here is that the stretch script isn't that well optimized (it's trying to change g-code in python, so it's going to be slow at best!). So what you're seeing is that the script is working through the g-code, but it's simply taking an insanely long time to do so, which freezes the UI while it's working.

ryanshow, 2019-10-03:
Ah, yep. You're exactly right. After viewing the logs and plotting a few times, it would appear that the time it takes the post stretch script to process a layer can be expressed in polynomial time...

[image]

So while Cura might technically not be crashing, I think my original problem still stands. It's reasonable to expect a script like this to finish it's job in a matter of minutes (with some sort of indication that it's working in the UI), not days for complex setups.

nallath, 2019-10-04:
The script in and out of itself is a bit hackish. The one thing we could add is that the interface is a bit more gracefull when doing it, but that's about it (eg; we simply can't fix the issue that the script might take days to complete)

Ghostkeeper, 2019-10-07:
I'd expect that script to be linear in the number of layers, but it could well be quadratic on the number of lines in a layer since it does collision checking.

closed 2024-12-07 by github-actions[bot]