Ultimaker/Cura #6469: Bumps other USB Arduino stuff

FlailAway, 2019-10-02:
Same thing I reported back in April
https://github.com/Ultimaker/Cura/issues/5613

**Application version**
4.3.0.BETA, latest Appimage

**Platform**
Mint + Cinnamon

**Printer**
Any

**Reproduction steps**
Have an Arduino IoT application running that uses the USB-Serial to transfer continuous data stream to PC.
Start Cura

**Actual results**
Crashes my data stream and I have to reestablish the connection.

**Expected results**
Not crashing my data stream and having to reestablish the connection.

**Additional information**
The last time I disabled all of the USB stuff in Marketplace and it solved the issue. I have done this and more, but this being Beta I suspect it is poking around with the USB for something during development.

Really irritating to come home and find the IoT (including security cameras) has been down for 6-hours because I did a slice for a long print before leaving for work and forgot to check the IoT was back up. I have an auto-restart function, but sometimes Cura will not release the USB assignment.

Is there anything I can do to stop this happening?

smartavionics, 2019-10-02:
> Is there anything I can do to stop this happening?

How about providing your cura.log file so we can see what plugins are loaded, etc. ?

fieldOfView, 2019-10-02:
Did you check is "all of the USB stuff in Marketplace" (really, it is just one plugin) is still disabled? There are a couple of environment variables that you can set to limit what ports are queried. See https://github.com/Ultimaker/Cura/blob/4.3/plugins/USBPrinting/USBPrinterOutputDeviceManager.py#L122

I am working on an alternative USB implementation based on the underlying code of Printrun as a Cura plugin. This will not do auto port discovery, and only use the port you select for a printer. It is not fully functional yet, but you can have a look here:
https://github.com/fieldOfView/Cura-SerialConnection

FlailAway, 2019-10-03:
I think I see the culprit, Line 185, just a guess though.
[cura.log](https://github.com/Ultimaker/Cura/files/3687367/cura.log)

FlailAway, 2019-10-03:
All USB and then some. :)

> https://github.com/fieldOfView/Cura-SerialConnection
"Automatically discovering printers over USB on each startup and during operation is a bad idea"

You got that right, a crazy "default" idea. I will try the plug in as I do not need it to work per-se, just want it to stop it tickling the USB/Serial port.

Thanks

FlailAway, 2019-10-03:
> https://github.com/fieldOfView/Cura-SerialConnection

How do I install it? I unzipped and added it to the home/.local/share/cura/4.3/plugins but it did not show up in the list.

The working PC is not on the Internet and it complained about not being able to connect to the Database and amusingly kicking off the IoT stuff -- again. :)

Ghostkeeper, 2019-10-07:
Looks like you can install that by downloading the [zip](https://github.com/fieldOfView/Cura-SerialConnection/archive/master.zip), extracting the folder to `~/.local/share/cura/plugins` and renaming the folder to `SerialConnection` (rather than `SerialConnection-master` or something). In the end the file `~/.local/share/cura/plugins/SerialConnection/plugin.json` should exist.

FlailAway, 2019-10-07:
> renaming the folder to `SerialConnection` (rather than `SerialConnection-master`).

Ahhh, OK, thanks that did it. It now shows up in Marketplace|Installed and I will see if it solves the problem for the Serial-bumps.

Ghostkeeper, 2019-12-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/5207.

closed 2019-12-06 by Ghostkeeper