Ultimaker/Cura #6470: [4.2.1] Z hop doesnt work while combing

smejfi, 2019-10-02:
**Application version**
<4.2.1>

**Platform**
Windows 10

**Printer**
CR10-5s

**Reproduction steps**
I didnt want to solve stringing issues, so I tried to check "combing mode" in Cura, it actually helped a lot, but especially in case of bigger prints, the nozzle keeps hitting the infill a little, so I checked also "Z hop when retracted", but it seems that it doesnt have any influence, Z hop is just not happening when "Combing" is turned on, whet I turn it off, it starts working again.

**Actual results**
Z hop is not happening when "Combing" is on although "Z hop when retracted" is checked

**Expected results**
I would expect that Z hop will be optional (and working) even in combing mode, I have read some post regarding some older version, where someone was asking for this feature and people were asking him why would he want to use such a combination, well, easy answer, to prevent nozzle hitting the infill when traveling above it ...

Ghostkeeper, 2019-10-03:
Yeah, I could reproduce this in Master as well.

smartavionics, 2019-10-03:
> Yeah, I could reproduce this in Master as well.

But not using z-hop while combing (within a single part) has always been Ultimaker's policy. Using z-hop while combing has been requested before and nothing was changed.

Ghostkeeper, 2019-10-07:
Yes, if a travel move stays within a single part it shouldn't retract and then also shouldn't perform a Z hop. When a travel move is made from one part to another though it'll have to cross through a wall and then we will retract (if retract is enabled). In this case we'd also have to hop.

Due to the newer Max Comb Distance With No Retract it gets more confusing though. I believe it then retracts but still does the detours for the combing. So we could Z hop as well then but that means that we'd be making the detour up in the air where the walls are not being hit anyway, so we might as well take the eagle's flight to the destination. But then our travel move could again be so short that we don't need to retract again which leads to circular reasoning. So in the end, it's probably better to ignore the Z hop for that case.

Goodfeat, 2019-12-26:
Is it possible to make the lifting of the Z axis always occur when moving
I encountered a problem when moving the nozzle at high speed.

[image]

This picture shows part of the code that must be executed with a "Z-Hop Height". But this does not happen
Although setting "max comb Distance With No Retract" = 4

Actually retraction does not need to be done to me
But the program does not allow me to enable the "Z-Hop Height" parameter without the retract setting enabled.
It is necessary that the extruder moves with a raised nozzle if the comb distance> 4mm

Ghostkeeper, 2019-12-30:
In the case of your picture, you can make it perform a Z hop by disabling combing (set it to "off"). It'll move directly over everything instead of trying to move inside the walls and such.

We currently don't have a way to do a Z hop without a retraction. In general it would be good to reduce the dependency between the Z hop, Retraction and combing routines. It's quite a big mess of settings dependent on each other and only working in certain cases but then having exceptions upon exceptions upon exceptions.

rburema, 2020-01-21:
Devs: Seems we forgot to link back to the internal ticker nr. before now: It's in progress as internal ticket CURA-6860.

mahtDFR, 2020-01-24:
This will be fixed in the next release.

closed 2020-01-24 by mahtDFR