Ultimaker/Cura #6471: 4.3     Z Seam bug

lwbsbde, 2019-10-03:
If I choose "User Specified" , I can not choose the Z Seam position.

nallath, 2019-10-03:
I can't reproduce this (although i did test this in master). I did the following:
1. Add UM3
2. Set "Z Seam Aligment" to 'User Specified' 
3. I can select values in the "Z seam X" and "Z seam Y" setting as well as "Z Seam Position"

lwbsbde, 2019-10-03:
in Cura 4.2.1  it worked too.  after update to 4.3 I have the error

lwbsbde, 2019-10-03:
[image]

nallath, 2019-10-03:
Try and also enable the "Z seam X" and "Z seam Y". You probably manually set a value to them. Once you do that, you can't use the overarching (parent) setting anymore

jcwebber93, 2019-10-03:
I have a similar issue, although it is slightly different depending on the printer selected. In all cases I have the following options selected as visible:

- Z Seam alignment
- Z Seam position
    - Z Seam X
    - Z Seam Y
- Seam Corner Preference
- Z Seam Relative

For UM3, When selecting "User Specified" for "Z Seam Alignment", I can use the "Z Seam Position" dropdown (back, back left, etc). I do not seem to have the option to select the X or Y like in 4.2.1.
[image]

For a custom machine, selecting "User Specified" for "Z Seam Alignment", I can see but cannot use the "Z Seam Position" dropdown, (grayed out), and in addition I do not seem to have the option to select the X or Y like in 4.2.1.
[image]

Visible settings for both test cases
[image]

smartavionics, 2019-10-03:
I have had similar problems with the X and Y coordinates not being visible even when they should be and I got around it by closing the infill section and then re-opening it. I think it's a QML weirdness.

jcwebber93, 2019-10-03:
> I have had similar problems with the X and Y coordinates not being visible even when they should be and I got around it by closing the infill section and then re-opening it. I think it's a QML weirdness.

Closed all sections while having the custom printer selected, and it added back in the X and Y locations that were previously hidden.  Thanks for the workaround!
[image]

lwbsbde, 2019-10-03:
Thanks.
 it works

Ghostkeeper, 2019-10-07:
I'm unable to reproduce that problem. Both UM3 and Custom FFF Printer just show the settings and allow me to change them. Maybe you could provide a project file saved when it was happening?

Maarten88, 2019-10-10:
I had the same problem, solved it using the tip in this issue: collapse the settings in and out, and the "Z seam X" and "Z seam Y" items re-appeared.
For reproduction I'd assume this probably happens only after migrating a profile having Z seam alignment set to "User Specified", with X and Y set to some value, from Cura 4.2 to 4.3.

Ghostkeeper, 2019-10-14:
I can't get the Z seam settings to disappear even when importing a profile made in 4.2 that contains those settings. This is the profile that I imported: [zseamprofile.zip](https://github.com/Ultimaker/Cura/files/3723798/zseamprofile.zip)

closed 2020-06-19 by no-response[bot]