Ultimaker/Cura #6472: Support Density Locked

B14d3r11, 2019-10-03:
I am attempting to change my Support Density from 20% back to 15% but it is greyed out and I can't touch it at all. I am having LOADS of problems with my prints snapping because the supports are too thick to be removed... even my flush cutters are having trouble getting through

Ghostkeeper, 2019-10-03:
Do you have a project file to share with us that reproduces the problem? When it slices with stupid thick supports, press File -> Save... and zip the resulting file, then upload it here. We may be able to point out a particular setting that's causing this then.

B14d3r11, 2019-10-05:
[Pan Hooks.zip](https://github.com/Ultimaker/Cura/files/3693652/Pan.Hooks.zip)

I printed these and it was fairly difficult to snap off the supports. I kept the same settings before the 4.0 to 4.3 upgrade I did

Ghostkeeper, 2019-10-07:
That's not a project file but the g-code. I can't see all your settings then. I do see that you've enabled Conical Support and gave it a minimum diameter of 25mm though. That's something you might want to disable since it's not something you're using in this print.

B14d3r11, 2019-10-07:
Yeah sorry, I misread what you said. Heres the project file (I think?)
[Project.zip](https://github.com/Ultimaker/Cura/files/3697988/Project.zip)

Ghostkeeper, 2019-10-08:
That's a project file! It's not made with the same settings as what you were using for that g-code though.

Ghostkeeper, 2019-10-10:
Ah, it's weird. I have to import the project twice in order to get it to load PLA - Coarse. Buggy there. And now it also has conical support enabled which I saw from your g-code.

Ghostkeeper, 2019-10-10:
It doesn't look particularly fat to me.

[image]

Maybe you mean that the Support Wall Line Count is set to 1? If you set that to 0, does that solve your issue?

B14d3r11, 2019-10-12:
The supports are thicker now than they were before. When it prints they were just, thinner. And I am using the same settings from 4.0 to 4.3 just Cura changed...

Its just harder to remove supports now than it was before

I am about to print something so i can try to show you what the supports look like now

Ghostkeeper, 2019-10-14:
Maybe you need to reduce the support line width? Is that what you mean?

closed 2020-06-19 by no-response[bot]