Ultimaker/Cura #6475: Restore use of Delete Key on the Mac

alsmithson, 2019-10-03:
On the Mac, Cura treats the delete key in a non-standard way. PrusaSlicer, IdeaMaker, Fusion 360, Microsoft Office, and every other Mac program I know of, use the Delete key to delete objects. Cura requires Fn+Delete. 

Please allow Mac owners to once again use the delete key to delete objects. This repository has several similar complaints/requests ( e.g., #3257, #2735, #237), and several times the functionality has been implemented, but it does not work in Cura 4.2.1 or 4.3.0 (the only versions I've used).

Fn+Delete works, but it is non-standard and affects workflow for power users. Mac-only users don't even know about Fn+Delete.

All Mac users will benefit from this change.

Ghostkeeper, 2019-10-03:
Just pressing the delete button should work (without Fn) as far as I know. However Apple keyboards do something weird with that button so that it behaves differently, which is why we had to bind different keys to do the same thing. I consider this a regression.

alsmithson, 2019-10-03:
It should work, but it does not in Cura. It works correctly in IdeaMaker, PrusaSlicer, and all other programs.

Ghostkeeper, 2019-10-07:
Something very funky is going on in MacOS here.

I went to https://keycode.info/ which tells me which buttons I'm pressing. These are the results from that website:

* Apple keyboard on Ubuntu: Delete button gives 8 (backspace).
* Apple keyboard on Ubuntu: Fn+Delete button gives 43 (delete).
* ISO9995 keyboard on Ubuntu: Backspace button gives 8 (backspace).
* ISO9995 keyboard on Ubuntu: Delete button gives 43 (delete).

So far, so good. In Cura, all 4 buttons work. Now for MacOS:

* Apple keyboard on MacOS: Delete button gives 8 (backspace).
* Apple keyboard on MacOS: Fn+Delete button gives 43 (delete).
* ISO9995 keyboard on MacOS: Backspace button gives 8 (backspace).
* ISO9995 keyboard on MacOS: Delete button gives 43 (delete).

Seems fine on that website. It's the same. However in Cura, the first one for MacOS (delete key on apple keyboard) doesn't work. The ISO9995 keyboard does work with both delete and backspace and Fn+Delete still works.

@LipuFei you had a development environment set up for MacOS? Is it possible to see which keys are being pressed there?

It seems to me like MacOS is processing the delete key differently for Apple keyboards and giving the application a different key, but maybe Firefox and Chrome are translating that for Javascript to the correct backspace key.

LipuFei, 2019-10-08:
I tested Cura 4.3 on my laptop with MacOS 10.14.6. Both `delete` and `Fn + delete` work for me.

@alsmithson 
 - Which OS version do you have?
 - What keyboard settings do you have?

This is my keyboard input source:
[image]

Ghostkeeper, 2019-10-08:
It was also occurring on the Macbook of @Yian-Lai.

alsmithson, 2019-12-14:
@LipuFei, I am using the US Keyboard layout for the keyboard built into my 2017 Macbook Pro running OS 10.14.6. It looks like the one in your attachment. As an experiment, I added the ABC keyboard shown in your screenshot, but I still can't use the delete key with Cura.

I brought up the keyboard viewer (virtual keyboard), and the delete key on it does delete an object in Cura, but the real delete on the real keyboard does not.

Ghostkeeper, 2019-12-16:
Which key code is pressed for you if you press the delete key on https://keycode.info/ ?

alsmithson, 2019-12-17:
I get backspace 8 when I press the delete key on https://keycode.info/ ?

I started noticing yesterday that once in a while the delete key does actually delete, though most of the time it does not. I did some experimenting tonight; I can consistently open a file and immediately delete it with the delete key. But if I work on that file by slicing, previewing, etc., then it is a crapshoot if the delete key works. 

I can't find a pattern, though. One time, I open a file in prepare view, slice it, then switch to preview mode, and so on, then it won't delete. But I repeat the experiment and it then it will delete.

Ghostkeeper, 2019-12-17:
I made a fix that it accepts keycode 8 from the Qt action. I think something may be loaded later that collides with the keyboard shortcut, but this should fix it.

commit bec513aa (2019-12-17): Alternatively allow backspace to delete selection

I thought we already had this?

Fixes #6475.
closed 2019-12-17 by Ghostkeeper