Ultimaker/Cura #6477: [4.3] Dual extrusion with single hotend shuts off heat at last extruder change.

Guanu, 2019-10-03:
**Application version**

Cura 4.3

**Platform**

Windows 10

**Printer**

Custom FFF printer with 2 extruders.

**Reproduction steps**

Machine with 2 extruders and one hotend using a Y adapter.  When an object uses 2 colors and has is no longer using both extruders, it will auto-insert a temperature shut off after the final extruder's start gcode section is inserted.  Since it is after the Start g-code of the final extruder used, it will shutoff the heat completely since it is a single hotend.  This prevents a user from adding the line M104 S{material_print_temperature} to the Start gcode of the extruders that would allow the temperature to continue.
[image]
In the above picture, Blue is extruder 1, Grey is extruder 2.  The dual extrusion ends at the top of the grey, and would continue with the blue remaining part of the benchy.
When this is sliced, after the first dual extrusion part is done, and only the rest of the benchy remains, Cura will send a temperature shut off after the final extruder is switched to.  If this line was between the extruder changes, it could be bypassed with a M104 S{material_print_temperature} command but since it is done after the final change, there is no way to bypass this. 
The temperature shut off is seen in the following screenshot:
[image]

**Actual results**

Auto inserted temperature shut off is inserted AFTER the switch to the final extruder. 

**Expected results**

Have the auto-inserted temperature shut off command be put in between the final extruder changes so the last extruder used can turn the temperature back on.

**Additional information**

Attaching Cura project file that can be directly sliced and the resulting gcode searched for M104 to find the temperature shut off command.
[Extruder-Shutoff.3mf.zip](https://github.com/Ultimaker/Cura/files/3688248/Extruder-Shutoff.3mf.zip)

smartavionics, 2019-10-03:
I have already submitted a PR that lets you tell Cura that the extruders share a heater. See https://github.com/Ultimaker/Cura/pull/5840 for details. I don't know if that PR will ever be accepted into Cura so if you want this feature you could try one of my builds that already have it (along with lots of other funky stuff). You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. They can be installed alongside the Ultimaker builds without zapping them.

nallath, 2019-10-04:
We have a reasonably empty sprint, so I expect we will (finally) have some time to do something about the older pull requests.

Guanu, 2019-10-04:
I just wanted to add that smartavionics build with the shared heater checkbox solves the problem completely, so if that can be pushed into the release it would be fantastic.  We instruct all of our customers to use cura, and this is a major roadblock for dual extrusion on our machines.  If this goes into a release this would fix a problem that has been around for over a year and would be perfect!

smartavionics, 2019-10-04:
Hi @Guanu , I'm pleased that my contribution solves the problem. But tell me this, if you have been having a problem with this for over a year, why didn't you (or your team) come up with a fix and contribute it yourselves? It's called give and take and that's how open source software works.

Guanu, 2019-10-04:
We don't have a software dev team and are a hardware manufacturer (and a small employee base), and I have made forum posts about the issue, emailed ultimaker multiple times and just kept getting a response that since it is not their machine there was nothing they would do.  I can edit some code, but not versed enough or have the software to edit the code.

Ghostkeeper, 2019-10-07:
Duplicate of https://github.com/Ultimaker/Cura/issues/4204

closed 2019-10-07 by Ghostkeeper